motioneye-project/motioneye #2245: Motion Eye Forgets USB Hard Drive

288Dave, 2021-12-07:
So I installed Motion Eye on a Pi 4 running buster and everything works fine! I can log into Motion Eye at (I.P. Address):8765. For media storage I'm using a 500 gb usb hard drive and was able to set it as the storage in Motion Eye. It works fine, until I reboot the Pi. Then Motion Eye forgets the preference for the usb drive and reverts to storing videos on the Pi's internal flash drive. What up with that? Any help appreciated. Thanks!

starbasessd, 2021-12-07:
Does the pi still show the drive attached? ssh or console:
df -h
Are you having to unplug it and re-plug the drive in to see it again?
How are you re-booting the pi?
Is it a physical platter HD? or USB thumb/flash? or usb adapter with m.2?

288Dave, 2021-12-07:
Thanks for the response! After a reboot the hard drive still shows on the desktop and with df -h. No need to unplug and replug. The Pi is being rebooted through the os. The hard drive is a platter hd. The weird thing is it worked perfectly when the Pi was running motion eye os. Since it's been over a year since that os has been updated I went with Buster. Motion Eye seems to run more smoothly when added to Buster, except it seems to forget the storage preference when the Pi is rebooted.

starbasessd, 2021-12-07:
How do you have the path set in File Storage?
The file that has that info is in /etc/motioneye/camera-X.conf files
Does it keep other settings?

288Dave, 2021-12-07:
I set the file storage path by right clicking on a camera's storage folder on the external hard drive, go to "copy path", then pasted that info into the Motion Eye admin page as a custom path for each camera's storage folder. Save the changes in Motion Eye and it works fine, until I reboot. All other setting seem to be remembered fine.

starbasessd, 2021-12-07:
What options do you get in the "Custom Path" drop down box?

288Dave, 2021-12-07:
So I set the path to SDA1 and the root directory to /camera1. Works fine. After a reboot the root directory changes to /media/pi/95dfb103-1391-4635-9fe2-6812f86e6a6b3/camera1. ?

starbasessd, 2021-12-07:
But is it recording to the correct folder?

starbasessd, 2021-12-07:
Which Buster are you using? RaspberryPiOS? If so which build? 5/2021 or 11/21?
Or the Real Debian Buster?

288Dave, 2021-12-07:
No. The file storage amount available drops from about 465 gigs to 58 gigs.

starbasessd, 2021-12-07:
How are you mounting the USB drive?

288Dave, 2021-12-07:
Raspberry Pi os buster, downloaded today. It said something about being the legacy version. How could I view this os info on the pi?

288Dave, 2021-12-07:
I just plug the usb drive into the pi and the os recognizes it. Do I need to change a setting to have it "mount" at every restart? It seems to do that anyway.

288Dave, 2021-12-07:
Running cat /etc/os-release on the pi shows buster version 10.

starbasessd, 2021-12-07:
Don't know off the top of my head, but there are some big differences between the Legacy (11/21) version and the previous (5/21) versions. You can still get the 5/21 version
[Lite](https://downloads.raspberrypi.org/raspios_lite_armhf/images/)
[Full](https://downloads.raspberrypi.org/raspios_full_armhf/images/)
(One of the changes they made was support of PiZeroW2 in the Legacy which wasn't supported in the 5/21 image)

288Dave, 2021-12-07:
Do you mean 5/28 version? I don't see a 5/21.

starbasessd, 2021-12-07:
May 2021...

288Dave, 2021-12-07:
I'll try that. Thanks.

288Dave, 2021-12-08:
Tried the May version of 32 bit Buster. Motion Eye remembered the external hard drive through a couple of reboots, then went back to forgetting it. Put motioneye os back on the Pi and everything is back to normal. The external hard drive is stable. Will there be any updates to motioneye os in the future? Maybe even 64 bit? Thanks.

starbasessd, 2021-12-08:
Doubtful for a 64bit motionEyeOS, heck, even RPiFoundation doesn't have a stable version of RPiOS... motionEye should install on any of the Arm64 on Pis.
Got to thinking. RPi4 has/had issues with platter HDs spinning up fast enough to register with the OS, to the point they put built in delays in the rpi-eeprom boot firmware. Check out:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#USB_MSD_DISCOVER_TIMEOUT
and other discussions:
https://raspberrypi.stackexchange.com/questions/33037/how-to-add-delay-at-boot
Although most are discussing booting from a USB drive, several are talking about the drive not mounting immediately, that they have to go through a 'mount -a' command in crontab with a delay to make sure the drive had enough time to spin up. It could be that your drive doesn't reset properly for a reboot (power not off long enough) or gets confused and doesn't allow a re-connect in time. That will be harder to diagnose. Most also note it can be drive specific and / or manufacturer specific. Most of my external 2.5" platter drives are matsunichi USB 3.0s from Microcenter, and I've not noticed a problem with them, even booting from them on 3Bs, 3B+s and 4Bs.

288Dave, 2021-12-08:
You're probably right about the drive. It's an older Seagate that I had lying around. Anyway, motioneye os seems to have no problem dealing with it. Thanks again!

jariojose, 2022-04-09:
I have the same problem. I think that @starbasessd  comment is the main issue for this problem. 

So, I think that is easily solved with adding a delay in motioneye service (/etc/systemd/system/motioneye.service):

`ExecStartPre=/bin/sleep <#ofseconds>.`

Old:

`[Unit]
Description=motionEye Server

[Service]
ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
Restart=on-abort

[Install]
WantedBy=multi-user.target
`

New:

`[Unit]
Description=motionEye Server

[Service]
ExecStartPre=/bin/sleep 10
ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
Restart=on-abort

[Install]
WantedBy=multi-user.target
`

I have no time to do this now. But, I think that it could work.
If your external drive won't is mounted automatically, could you add some scripts to mount it and them, start the motioneye service.

MichaIng, 2022-04-09:
A random delay may be a workaround for a particular case, but so solution. The recent dev motionEye systemd unit already waits for network. Probably adding `After=local-fs.target` helps to additionally delay the service start to after all local filessystems have been mounted or the automount units initialized. `remote-fs.target` allows the same for network drives, like NFS and CIFS mounts.

jariojose, 2022-04-09:
Nice!! I will do this! Thank you!

MichaIng, 2022-05-02:
I had a look into the motionEyeOS image, especially the init.d services:

[image]

AFAIK SysV start services one by one instead of in parallel, and based on the numerical service name prefixes, `mountall` starts before `motioneye`. With `After=local-fs.target remote-fs.target` we should be able to achieve the same for the motionEye systemd service. I'll open a PR to add it like that. Better would be to delay drive detection and media file writes until the respective mount has been done, while allowing the service itself to start earlier, but this is much more complicated to implement, at least with my still limited insights.

MichaIng, 2022-05-02:
PR up: #2452

This change aligns it with the SysV service as well: https://github.com/motioneye-project/motioneye/blob/ad6f69a/motioneye/extra/motioneye.sysv#L7

MichaIng, 2022-05-04:
Merged into `dev`.

closed 2022-05-04 by MichaIng