motioneye-project/motioneye #2246: Device Reboots every few seconds. Can't seem to find a reason from logs

hmxmghl, 2021-12-08:
Hello,

I am running motionEyeOS 20200606 on Raspberry Pi 3B. Device reboots every few seconds. I have tried exploring the logs but unable to find anything related to it. In logs from previous days, it shows that interrupt signal is causing the reboot but it's not mentioned which interrupt was it or what caused the interrupt? In present day logs, there is no information of interrupts at all.

I have tried many things:
- Increase timeout for Network Link Watch
- Disable Network Link Watch
- Disable all unused Services
- Increasing GPU Memory

**Logs**
[motioneye.log](https://github.com/ccrisan/motioneye/files/7673666/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneye/files/7673667/messages.log)
[boot.log](https://github.com/ccrisan/motioneye/files/7673668/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneye/files/7673669/dmesg.log)

Also, there is a secondary issue. When using Google Driver service to upload pictures to Google Drive, it sometimes fails to resolve the name of the file and crashes and reboot.

Any help would be highly appreciated. Thanks!

Hamza R.

starbasessd, 2021-12-08:
Interesting logs. 
I would clean re-image, without any changes (other than adding cameras), using dev20201026.
I saw something about a GPS, If you have a hat, I'd remove it, too. 
It's weird. I don't see any reports of the system temperature. I don't see any reports of a 'panic' (watchdog shutting it down). It's as if the power gets disconnected. I would also check your power supply, as a start.

hmxmghl, 2021-12-09:
Nice catch with the power supply issue. Logs show no trace of anything happening within SW. Device just reboots out of nowhere so probably it is power issue. I have ordered a new supply now and will be testing with it. Will update here. Thanks!

hmxmghl, 2021-12-11:
Changing the power supply and clean re-imaging and fixed the issue. Closing now!

closed 2021-12-11 by hmxmghl