motioneye-project/motioneye #2247: Not possible to add an additional USB camera

perlian, 2021-12-09:
I need help

I have actual version 042.1 running
I tried different USB Cameras
All of them where detected as local v4l2 camera but only a grey screen is shown. 
so no USB camera is working obviously
Internal Raspberry camera module is working fine

starbasessd, 2021-12-09:
@perlian Please provide the following info:
Board being used
OS being used
Which set of instructions to set up motionEye being used
Which version of motion being used
Which USB cameras being used
We will be glad to try to help.

perlian, 2021-12-09:
I use the Image for Raspberry Pi on a Raspberry Pi 3
Version is 0.42.1
I tried different Logitech Webcams

starbasessd, 2021-12-09:
motionEyeOS?
In either, go to the Settings, go down to General
What does it say for:
Motion version?
OS Version?
Also on a Raspberry Pi (and most SBCs (single board computers)) You can only have 1 USB camera. It's a limitation of the bandwidth available on the USB bus. On some PCs you can have multiple USB cameras, but only one per USB Controller.
If you have more than one camera, unplug all but 1 and see if it works.

Choengboy, 2021-12-14:
Help me

starbasessd, 2021-12-14:
@Choengboy Help you with what?

JamBalaya56562, 2026-11-08:
Closing, as this got no answer to the versions asked for and was with the motionEye 0.42.1 image on a Pi 3. USB webcams that are detected but stay grey there, while the camera module works, usually hit the bandwidth of the Pi 3's single USB bus, as starbasessd explained; a lower resolution and frame rate, or one USB camera at a time, is the way to check. Please open a new issue with `motion.log` if a USB camera still stays grey with a current release.

closed 2026-11-08 by JamBalaya56562