motioneye-project/motioneye #2248: Camera field in New Camera add is not a dropdown

jtmpush18, 2021-12-17:
Hello.
I just installed Motioneyeos on a Odroid HC4.
motionEye Version | 0.42.1
Motion Version | 4.3.2
OS Version | Debian 11

I'm really new at this, but I think I did the install correctly.
I clicked on add a new camera, and I got the dropdown.
It's a network camera, an Amcrest IP3M-HX3.
I added the URL (rtsp://[username]:password@IPaddress:554/cam/realmonitor?channel=1&subtype=0)
I added the id and password.
However, the last box, called "Camera" doesn't have any dropdown.
What am I doing wrong?

[image]

starbasessd, 2021-12-17:
Which image are you using for motionEyeOS? The HC4 isn't a listed supported version.
'Normally' you would use either
rtsp://[username]:password@IPaddress:554/cam/realmonitor?channel=1&subtype=0
with nothing in the Username & Password boxes (not recommended) or
rtsp://IPaddress:554/cam/realmonitor?channel=1&subtype=0
and Username & Password in the appropriate boxes.
If nothing shows in the Camera box, it means it didn't find anything for that URL.
I looked at the iSpy database, and didn't see that model listed. 
https://www.ispyconnect.com/camera/amcrest
It may be that it uses proprietary encoding, requires a cloud connection, or other issues.
Searched for 'IP3M-HX3' (none) 'IP3M' (lots) 'HX3' (none).
Does your URL work with VLC?

jtmpush18, 2021-12-17:
Thank you.
I took the id & password out of the url, and I got success.

It is an Amcrest IP3M-HX2W, and I've got an image!
On to the next challenge.
Thanks again.
P.S. I know the Odroid-HC4 isn't a supported box, but it seems to work....
As mentioned above it's running Debian 11 (armbian).

jtmpush18, 2021-12-17:
P.P.S. If you want me to run any tests or betas on the Odroid-HC4, no problem.

closed 2021-12-17 by jtmpush18

starbasessd, 2021-12-17:
If you are running an OS (armbian) Debian Bullseye did you follow the motionEye installation instructions here:
https://github.com/ccrisan/motioneye/wiki/Install-on-Debian-11-(Bullseye)?
In which case you have motionEye, not motionEyeOS...