motioneye-project/motioneye #2249: Motion Detection - involving Aperture time to Motion Detection

pgreendale, 2021-12-19:
Camera : NOIR V1 
RPi: 3A+ 

Spend some time in working out why motion detection doesnt work on my setup, tried all possible settings with filtering, auto modes, , lightswitch detection off, thresholds etc but it just didnt cause action. Logfiles showed motion daemon is working fine, and also frame change detection showed results so I went further. 

Lately I found the cause. My Camera is running hardware auto brightness, causing high exposure times in the 0.5s-1s range. Combined with the "Minimum Motion Frames " set to default 20, that means 10 seconds of movement above thereshold from frame to frame which is far too slow for my intended surveillance application. 

I suggest an option which shows the actual necessary motion timespan and/or  lets the user set a specific timespan which in between some movement has to be detected for causing action. (like time/exposure based calculation of frames with presettable minimum  frame when exposure time exceeds intended timespan). 

Aside, the other pieces are working fine, great work!

starbasessd, 2021-12-19:
That is what the whole section: Settings, Motion Detection is all about.
Motion Gap is minimum time between valid motion detections
Minimum Motion Frames is the minimum number of consecutive frames with differences to be a valid detection.
Motion/MotionEye doesn't care how long a frame takes, whether it's 10FPSecond or 10 FPHour
In Settings, Video Device, try turning off "Auto Brightness" and increase the Frame Rate to something around 10-15 FPS...
In my screenshot, these are my usual camera starting settings...
[image]

pgreendale, 2022-08-10:
Thanks, thats it!

closed 2022-08-10 by pgreendale