motioneye-project/motioneye #2250: Install on Debian 11 fails

Fuxameralu, 2021-12-22:
newly installed Debian System, Version 11.2, without desktop environment

I followed the Install Instructions to Point 4, Installation of motioneye, where the Installation fails with following Output. Unfortunately, I have not found a solution myself an would appreciate help. Thank you

 **sudo pip install motioneye
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Collecting motioneye
  Using cached motioneye-0.42.1.tar.gz (758 kB)
Collecting tornado<6,>=3.1
  Using cached tornado-5.1.1.tar.gz (516 kB)
Collecting jinja2
  Using cached Jinja2-2.11.3-py2.py3-none-any.whl (125 kB)
Collecting pillow
  Using cached Pillow-6.2.2-cp27-cp27mu-manylinux1_x86_64.whl (2.1 MB)
Collecting pycurl
  Using cached pycurl-7.43.0.5.tar.gz (216 kB)
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-PxaYTp/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-PxaYTp/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-l2fIpA
         cwd: /tmp/pip-install-PxaYTp/pycurl/
    Complete output (11 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-PxaYTp/pycurl/setup.py", line 961, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/tmp/pip-install-PxaYTp/pycurl/setup.py", line 623, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-install-PxaYTp/pycurl/setup.py", line 101, in __init__
        self.configure()
      File "/tmp/pip-install-PxaYTp/pycurl/setup.py", line 238, in configure_unix
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.**

starbasessd, 2021-12-22:
Starting a new install:
Debian 11, Server, full update to 22 Dec 2021 (VBox VM, 4GB, 50GB, 4 Cores)
Will update as needed.

starbasessd, 2021-12-22:
VBox VM: 4GB, 50GB, 4 Cores.  Debian 11 (without GUI Desktop), Full update to 22 Dec 2021. 
su -
apt-get install motion ffmpeg v4l-utils -y
systemctl stop motion
systemctl disable motion
apt-get install python2 curl -y
curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py
python2 get-pip.py
apt-get insmkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
mkdir -p /var/lib/motioneye
tall python-dev-is-python2 python-setuptools libssl-dev libcurl4-openssl-dev libjpeg-dev zlib1g-dev libffi-dev libzbar-dev libzbar0 -y
pip install motioneye
mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
mkdir -p /var/lib/motioneye
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye
Note: On a clean new install, nothing else installed, you need curl to pull pip2 (noted above and fixed in documentation) which must be installed prior to that step. Other than that, it installed correctly and ran at http://ip_address:8765

starbasessd, 2021-12-22:
I didn't mean to close this issue.

Fuxameralu, 2021-12-22:
I'll try it out tomorrow (cet time zone here) and give you feedback. Thanks so far for the quick response

starbasessd, 2021-12-22:
Would have been faster, but couldn't clone a base image for some reason, so had to build from scratch.

Fuxameralu, 2021-12-23:
Problem solved. Thanks alot

closed 2021-12-23 by Fuxameralu

meegle84, 2022-03-27:
> VBox VM: 4GB, 50GB, 4 Cores. Debian 11 (without GUI Desktop), Full update to 22 Dec 2021. su - apt-get install motion ffmpeg v4l-utils -y systemctl stop motion systemctl disable motion apt-get install python2 curl -y curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py python2 get-pip.py apt-get install python-dev-is-python2 python-setuptools libssl-dev libcurl4-openssl-dev libjpeg-dev zlib1g-dev libffi-dev libzbar-dev libzbar0 -y pip install motioneye mkdir -p /etc/motioneye cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf mkdir -p /var/lib/motioneye cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service systemctl daemon-reload systemctl enable motioneye systemctl start motioneye Note: On a clean new install, nothing else installed, you need curl to pull pip2 (noted above and fixed in documentation) which must be installed prior to that step. Other than that, it installed correctly and ran at http://ip_address:8765

Thanks a lot!
Working in my RPI Debian GNU/Linux 10 (buster)