motioneye-project/motioneye #2252: Why is feature development stopped?

developerfromjokela, 2021-12-22:
@ccrisan

developerfromjokela, 2021-12-25:
@starbasessd if this is a duplicate question, please link the original issue.

starbasessd, 2021-12-25:
After so many requests for updates, I stopped.

starbasessd, 2021-12-25:
And, technically it's not 'stopped', Just that ccrisan hasn't posted any. Jawsper and Jasaw and others have posted updates, fixes, a new build, and there are several forks out there...

developerfromjokela, 2021-12-28:
> And, technically it's not 'stopped', Just that ccrisan hasn't posted any. Jawsper and Jasaw and others have posted updates, fixes, a new build, and there are several forks out there...

Could you link repos with active builds?

starbasessd, 2021-12-28:
I can't support Jawsper's build (I don't have an RPiZeroW2) but it's in thread 
https://github.com/ccrisan/motioneyeos/pull/2829
Do you expect every github project to cross link with every other forked project?

closed 2021-12-28 by starbasessd