motioneye-project/motioneye #2253: Installation failed Raspberry PI 4

Floatcher, 2021-12-24:
I just tried to install motioneye on my new installed pi 4. with iobroker. I used the guide from here.

When I run "apt-get install python-pip ..." I get the message that python-pip is not available and I should use python3-pip instead. This I can install. 

But now, when I want to start motioneye, I get these error messages in the log

Dez 24 08:47:34 raspberrypi systemd[1]: Started motionEye Server.
Dez 24 08:47:34 raspberrypi meyectl[2765]: Traceback (most recent call last):
Dez 24 08:47:34 raspberrypi meyectl[2765]:   File "/usr/local/bin/meyectl", line 5, in <module>
Dez 24 08:47:34 raspberrypi meyectl[2765]:     from motioneye.meyectl import main
Dez 24 08:47:34 raspberrypi meyectl[2765]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/meyectl.py", line 28, in <module>
Dez 24 08:47:34 raspberrypi meyectl[2765]:     import settings
Dez 24 08:47:34 raspberrypi meyectl[2765]: ModuleNotFoundError: No module named 'settings'
Dez 24 08:47:34 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Dez 24 08:47:34 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.
Dez 24 08:48:07 raspberrypi systemd[1]: Started motionEye Server.
Dez 24 08:48:07 raspberrypi meyectl[2809]: Traceback (most recent call last):
Dez 24 08:48:07 raspberrypi meyectl[2809]:   File "/usr/local/bin/meyectl", line 5, in <module>
Dez 24 08:48:07 raspberrypi meyectl[2809]:     from motioneye.meyectl import main
Dez 24 08:48:07 raspberrypi meyectl[2809]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/meyectl.py", line 28, in <module>
Dez 24 08:48:07 raspberrypi meyectl[2809]:     import settings
Dez 24 08:48:07 raspberrypi meyectl[2809]: ModuleNotFoundError: No module named 'settings'
Dez 24 08:48:07 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Dez 24 08:48:07 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.

Floatcher, 2021-12-24:
Solved it with installing pip2 and reinstalling motioneye.
Found here: 
https://giters.com/ccrisan/motioneye/issues/2117

Suggestion: Add the pip2 check to the installation instruction for Raspian. 

Thx!

starbasessd, 2021-12-24:
The instructions for debian and RaspberryPiOS already have the corrected pip2 install instructions here:
https://github.com/ccrisan/motioneye/wiki/Installation

closed 2021-12-24 by starbasessd