motioneye-project/motioneye #2254: Cant install on fresh Arch Linux setup

kmsgli, 2021-12-27:
Hey did a search with no luck, the Arch Linux install page no longer works as pip2 is not available to download.
I tried installing the motioneye AUR package but some of the dependencies no longer exist. I tried the manual install with no luck.

The motioneye-git AUR package installs but fails to start motioneye with some type of python3 error.

How am I supposed to install on Arch Linux?
Any help is appreciated.

starbasessd, 2021-12-27:
Can you use:
curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py
python2 get-pip.py
I am not familiar with Arch, I will look at it today.

starbasessd, 2021-12-27:
Arch makes Gentoo look simple. What install instructions do you use? I am now on attempt number 4 just to get the OS running in a VBox VM.

starbasessd, 2021-12-27:
Oh, boy. That is something that makes me appreciate Gentoo and Debian. 
I was able to finally get a good, clean install of Archlinux after a (large) number of attempts.
I then corrected / updated the Install instruction page for Arch here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Arch
Let me know if this resolves it for you.

kmsgli, 2021-12-27:
Full disclosure I have never used Gentoo but Arch gets a bad wrap, the AUR usually has everything you need and once you get used to it I would never switch away from it . The AUR does have a motioneye package that I used some years ago to install on my system and have never had an issues however it seems according to the AUR page "python2-pycurl-motioneye" is not a package anymore so it can not build.

https://aur.archlinux.org/packages/motioneye/

My system still works find because I have that package locally but I was setting up a new system for a friend and it seems moving forward you can not get that package to build it proper.

Your steps definitely solved the new install and I appreciate your help. I wonder if I can get whoever maintains that package to update the PKGBUILD to follow your instructions to automate the process. (maybe its time I give back and try to maintain it myself).

There is a motioneye-git in the AUR but that seems to be a work in progress using python 3? So maybe that will solve this issue down the road when that is complete (it builds but motioneye does not run).

Thanks again you have been a huge help!

starbasessd, 2021-12-27:
Glad to help. I doubt the package on Arch AUR is very current (could be, but doubt it) and none of the other distros has motionEye pre-packaged, AFAIK. That is why the Wiki instructions exist. Most of the distros have motion in them, but they are from when the distro was first set up and very outdated, or have some other issue. motion 4.4.x seems to have issues, that is why most of the Wiki instructions suggest 4.2.x or 4.3.x. Python3 is a 'whole 'nuther can o worms...

My references to Gentoo are because Gentoo literally builds everything from scratch and sources, and me and gcc are not the best of friends... <sigh> <grin> Arch feels like that even though it's mostly built packages instead of source code...

closed 2021-12-27 by starbasessd

kmsgli, 2021-12-27:
Well according to my package list I am on motioneye 0.42.1-3 that has been updating with the AUR package. I think that is fairly current although maybe I should migrate my system over to use the python2 setup, but honestly If motioneye on python3 is not too far away I would rather wait because someone is maintaining that package already and using pacman does not require me to remember to explicitly update motioneye.

Interestingly enough my version of motion is motion 4.4.0-1. I wonder if downgrading that will allow me to use the python 3 version of motioneye?

These are random thoughts now so I will take my win that you gave me and run, thanks again!