motioneye-project/motioneye #2255: issue with rtsp:// URIs that contain special characters like "."

HuguesD, 2021-12-27:
Hello

in the current (0.42) motioneye release, adding a RTSP (network) camera fails when the rtsp URL contains a dot (".") character in its address. For example rtsp://192.168.1.50:554/H.264 does not work. After trials and errors, i found that an alternative rtsp address for the same camera that does not contain the "." works fine.

This is a critical bug that needs correction because some cameras only have rtsp addresses containing ".". For example EZVIZ C6CN camera.

Is his bug known already ? Will it be fixed in future release ? When is the next release ?

Thanks
Hugues

starbasessd, 2021-12-27:
In the iSpy database there are numerous listings for rtsp
https://www.ispyconnect.com/camera/ezviz
with the C6CN showing with numerous possibilities.
My cameras use a dot in it's path without issues (D-Link  DCS-5020-L) without issue.
Did you try  rtsp://192.168.1.50/H.264 ? 
I don't see any on the iSpy database for your make and model that uses the port :554 in the the URL, they all seem to want to connect on the port 80 default.

zagrim, 2021-12-27:
I haven't looked into the code to see if there are any obvious reasons for this issue, but @HuguesD could you please try if substituting all dots in the URL (after the first `/`) with `%2E` helps? For example: `rtsp://192.168.1.50:554/H%2E264` instead of the URL you gave.

HuguesD, 2021-12-28:
> 
[image]
rtsp://192.168.1.50:554/H%2E264

Hi Zagrim, I tried your modified RTSP URL but no luck. It does not work still. See screenshot : no video device

HuguesD, 2021-12-28:
And additional info : the URL  : rtsp://192.168.1.53:554/H%2E264 works in VLC on the same computer.

starbasessd, 2021-12-28:
Did you try rtsp://192.168.1.50/H.264

cap9qd, 2021-12-28:
Is the camera stream 640x480? I've had issues in the past where meye wasnt
set for the right resolution and it would show the slashed box like you
have there. Just a thought...

HuguesD, 2022-01-01:
> Did you try rtsp://192.168.1.50/H.264

Yes I did, same result

starbasessd, 2022-01-01:
Did you check @cap9qd 's suggestion? What is the default resolution for the camera?

HuguesD, 2022-01-02:
> Did you check @cap9qd 's suggestion? What is the default resolution for the camera?

Yes, the camera is 1920x1080 native resolution. The stream (rtsp) is also 1920x&080. I tried different resolutions in motioneye but this does not change the outcome.

starbasessd, 2022-01-02:
I don't know. I am not able to reproduce the issue. I know there are issues with the high resolutions in motion/ffmpeg. I have 2 rtsp cameras with '.' in the URLs (Foscams) that work fine at 1280x720, 3 mjpeg (d-Links) with '.' that work fine at 640x480, so I am stumped.

HuguesD, 2022-01-16:
The parsing of the RTSP URL must be checked and corrected to not interpret the "." character as being not part or the URL string. I tested the same camera with an alternate RTSP URL that does not contain the "." and it works...

starbasessd, 2022-01-16:
Again, I am using 2 different Foscams with 2 different URLs that both have '.' in them (not as part of the IP Address) that both are parsed correctly. Do your logs show anything in them?

HuguesD, 2022-01-23:
Sorry, i decided finally to abandon this path.

closed 2022-01-23 by starbasessd