motioneye-project/motioneye #2256: Exposing admin username in javascript could be a potential security risk

developerfromjokela, 2021-12-28:
Hello!
motionEye seems to expose admin username in its HTML as javascript variable.
I think this is a potential security risk for brute-force and/or dictionary attacks towards the login system.

What was/is purpose of keeping that variable in HTML?

G2G2G2G, 2022-02-04:
This has nothing to do with brute force or dictionary attacks in any way.
data logging/sniffing/reading sure

developerfromjokela, 2022-02-04:
It has. If you don't expose your username, they cannot login even if they had correct password.

developerfromjokela, 2022-02-04:
It makes that job easier

MichaIng, 2022-03-19:
> What was/is purpose of keeping that variable in HTML?

It is the way how it is passed from the backend to browsers JavaScript to check whether the current user is the admin user, to show/hide enable/disable certain GUI elements:
- Set here via inline script, generated by backend: https://github.com/motioneye-project/motioneye/blob/c7d86c6/motioneye/templates/main.html#L89
- Use for this function in main JavaScript, executed in browser: https://github.com/motioneye-project/motioneye/blob/4249e44/motioneye/static/js/main.js#L398

However, I agree this is not so awesome. The admin username cannot be changed via GUI, but at least manually via config file. Since it is not used anywhere else in the frontend, it makes more sense to pass an "isAdmin" flag instead of the name. Even smarter would be to an admin or non-admin class to admin-only HTML elements right with the backend to have them shown/hidden in the first place, without needing to toggle classes via JavaScript, making even passing this flag obsolete.