motioneye-project/motioneye #2257: LICENSE file missing in pypi tarball

rustypig91, 2021-12-29:
I could not find any license text inside https://files.pythonhosted.org/packages/c5/ff/139b4b02ed089b2b50eee65d1059ae0d29064ab5f4ad429e45c702396697/motioneye-0.42.1.tar.gz

starbasessd, 2021-12-29:
License, as in is motionEye using a version of GPL or similar?

rustypig91, 2021-12-29:
The [LICENCE](https://github.com/ccrisan/motioneye/blob/dev/LICENSE) file could be included in the pypi package

MichaIng, 2022-03-12:
Makes sense. I guess @jmichault and/or @cclauss would know best how to achieve this. Else I can have a closer look when we are closer to a new stable release.

I first thought it may be best to clearly separate the actual code from meta files, including sample configs, service files etc, but probably it is even good to include all meta files into the resulting Python module, basically moving the content of `motioneye` sub directory into the parent? But if wanted, that should be done with case when ready pull requests  have been merged, as it causes a lot of conflicts.

MichaIng, 2022-03-12:
Solved in `dev` via #2346. But it may take a while until we do a new PyPI release. Marking this as closed.

PR #2346 (2022-03-12): Add LICENSE to setup.cfg
Closes #2257  @rustypig91
**GNU General Public License v3.0** as discussed in https://setuptools.pypa.io/en/latest/userguide/declarative_config.html
commit 10cedcc7 (2022-03-12): Add LICENSE to setup.cfg (#2346)

Closes #2257
GNU General Public License v3.0 as discussed in https://setuptools.pypa.io/en/latest/userguide/declarative_config.html
closed 2022-03-12 by MichaIng