motioneye-project/motioneye #2259: Grey Screen on Reolink RLC-810A

uwe72, 2021-12-30:
### motionEyeOS Version

I am running motionEyeOS version: 
motionEye Version | 0.42.1
Motion Version | 4.3.2

### Board Model
I am using the following board/model: Laptop Linux OS Version | Ubuntu 21.10. Intel i5

### Camera
Reolink RLC-810A

### Network Connection
My motionEyeOS unit is connected to the network via: Ethernet

### Log Files
I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 [motioneye.zip](https://github.com/ccrisan/motioneyeos/files/7794155/motioneye.zip)

### Problem
I tried many many variations of links, e.g.
rtsp://192.168.178.89:554/h264Preview_01_main
rtsp://192.168.178.89/h264Preview_01_main

But either it is a grey screen (unable to display video device) or it is black. Just the jpg stream is working

I updated already to the newest firmware of the camera.

[motioneye.log](https://github.com/ccrisan/motioneye/files/7794248/motioneye.log)
.

starbasessd, 2021-12-30:
Try disabling smarthome, and just run motionEye by itself to see if it lets the camera work.
If you installed motioneye as part of smarthome, uninstall smarthome and follow the instructions here to install motionEye:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
Then add and test your camera.

uwe72, 2021-12-30:
I shuted down smart home now. But everything same, works on VLC, but not on motion eye.
I installed motioneye using the instruction you posted.

[image]

uwe72, 2021-12-30:
`Dez 30 19:26:00 smarthome motion[36615]: [3:ml3:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normale Auflösung: Kamera kann nicht geöffnet werden ( Camera1): Server returned 401 Unauthorized (authorization failed)
Dez 30 19:26:00 smarthome motion[36615]: [3:ml3:Camera1] [ERR] [VID] vid_start: Netcam RTSP konnte nicht geöffnet werden
Dez 30 19:26:00 smarthome motion[36615]: [4:ml4:Camera2] [ERR] [NET] netcam_rtsp_open_context: Normale Auflösung: Kamera kann nicht geöffnet werden ( Camera2): Server returned 401 Unauthorized (authorization failed)
Dez 30 19:26:00 smarthome motion[36615]: [4:ml4:Camera2] [ERR] [VID] vid_start: Netcam RTSP konnte nicht geöffnet werden
Dez 30 19:26:01 smarthome motion[36615]: [1:ml1:Terrasse] [WRN] [NET] netcam_init_jpeg: kein neues bild, kein signal rcvd
Dez 30 19:26:01 smarthome motion[36615]: [3:ml3:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normale Auflösung: Kamera kann nicht geöffnet werden ( Camera1): Server returned 401 Unauthorized (authorization failed)
Dez 30 19:26:01 smarthome motion[36615]: [4:ml4:Camera2] [ERR] [NET] netcam_rtsp_open_context: Normale Auflösung: Kamera kann nicht geöffnet werden ( Camera2): Server returned 401 Unauthorized (authorization failed)
Dez 30 19:26:01 smarthome motion[36615]: [3:ml3:Camera1] [ERR] [VID] vid_start: Netcam RTSP konnte nicht geöffnet werden
Dez 30 19:26:01 smarthome motion[36615]: [3:ml3:Camera1] [WRN] [ALL] motion_init: Das ursprüngliche Bild konnte nicht von der Kamera abgerufen werden
Dez 30 19:26:01 smarthome motion[36615]: [4:ml4:Camera2] [ERR] [VID] vid_start: Netcam RTSP konnte nicht geöffnet werden
Dez 30 19:26:01 smarthome motion[36615]: [4:ml4:Camera2] [WRN] [ALL] motion_init: Das ursprüngliche Bild konnte nicht von der Kamera abgerufen werden
Dez 30 19:26:01 smarthome motion[36615]: [4:ml4:Camera2] [WRN] [ALL] motion_init: Die Bewegung verwendet weiterhin die Breite und Höhe der Konfigurationsdatei (en).
Dez 30 19:26:01 smarthome motion[36615]: [3:ml3:Camera1] [WRN] [ALL] motion_init: Die Bewegung verwendet weiterhin die Breite und Höhe der Konfigurationsdatei (en).
Dez 30 19:26:02 smarthome motion[36615]: [1:ml1:Terrasse] [WRN] [NET] netcam_init_jpeg: kein neues bild, kein signal rcvd
`

uwe72, 2021-12-30:
Seems i need to add login/password somehow

starbasessd, 2021-12-30:
With smarthome disabled and VLC closed:
Delete the camera
Re-add the camera with the same URL shown in the VLC window in the screenshot above.

starbasessd, 2021-12-30:
Are you adding the username and password for the camera in the username and password in the Add Camera box?
I don't understand the language your log is in, but I think its saying that you have the wrong username and password.

starbasessd, 2021-12-30:
Go to motionEye WebGUI http://ip_address_of_PC:8765 and add the cameras there.

uwe72, 2021-12-30:
Worked now like this:
[image]

THANKS A LOT!!

closed 2021-12-30 by starbasessd