motioneye-project/motioneye #2260: Only time outs when accessing the webinterface

manno-xx, 2022-01-01:
The webinterface is not accessible. The time outs happen every time.

**Hardware** is RPI Zero 2 W. 

SSH output
=
cat /etc/os-release
-
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

sudo systemctl status motioneye
-
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2022-01-01 17:53:24 GMT; 11min ago
 Main PID: 295 (meyectl)
    Tasks: 5 (limit: 718)
   CGroup: /system.slice/motioneye.service
           ├─295 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
           └─481 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

Jan 01 17:53:24 zero2wbirdcam systemd[1]: Started motionEye Server.
Jan 01 17:53:29 zero2wbirdcam meyectl[295]:     INFO: hello! this is motionEye server 0.42.1
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: cleanup started
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: wsswitch started
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: tasks started
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: mjpg client garbage collector started
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: server started

journalctl | grep -E "meyectl|motion"
-
Jan 01 17:53:24 zero2wbirdcam systemd[1]: Started motionEye Server.
Jan 01 17:53:25 zero2wbirdcam motion[302]: Not starting motion daemon, disabled via /etc/default/motion ... (warning).
Jan 01 17:53:29 zero2wbirdcam meyectl[295]:     INFO: hello! this is motionEye server 0.42.1
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: cleanup started
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: wsswitch started
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: tasks started
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: mjpg client garbage collector started
Jan 01 17:53:35 zero2wbirdcam meyectl[295]:     INFO: server started

I tried multiple ports, the default (8765) and 8080 by editing /etc/motioneye/motioneye.conf.
Port scans do indicate the ports are open, whichever is active at that time.

Another RPi running Domoticz runs fine (as have others) and is accessible on port 8080, so it cannot be some network issue (running everything on a mobile hotspot atm)

I initially tried the motionEyeOS fork from jawsper ([https://github.com/jawsper/motioneyeos/releases/tag/20211113](https://github.com/jawsper/motioneyeos/releases/tag/20211113)) but that gave me errors on the network device and it went into the panic reboot.

Any pointers on how to solve this?
Kind regards,
Manno

starbasessd, 2022-01-01:
Which Buster did you install? Legacy or previous version?
Which instruction set did you use to install motionEye?

manno-xx, 2022-01-01:
The OS is "Raspberry Pi OS Lite (Legacy)" Released 2021-12-02 as listed in Raspberry Pi Imager v1.6.2

For installation I used [https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)

Errrr, yes, maybe I should get hold of the Nov 8 release.

As a side note: is there any preference as to Buster over Bullseye? I figured the former bc of the new camera stuff etc. in Bullseye.

starbasessd, 2022-01-01:
No, Bullseye has all kinds of issues, especially with PiCams (CSI ribbon connector) but if you are using Network Cameras it can work as tested...
I will run a test with the 2021-12-02 image again (but on a Pi3B+ and wifi as I still haven't received my PiZeroW2 yet).

starbasessd, 2022-01-01:
Clean install RPiOS Lite Legacy (2021-12-02)
copy wpa_supplicant.conf & ssh to /boot partition
install to Pi3B+ and run 
> sudo apt update && sudo apt upgrade -y

set standard options via 
> sudo raspi-config

Locale, Time Zone, password, hostname
reboot
login

> sudo su
> apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y	

> wget https://github.com/Motion-Project/motion/releases/download/release-4.3.2/pi_buster_motion_4.3.2-1_armhf.deb

> dpkg -i pi_buster_motion_4.3.2-1_armhf.deb	

> apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev -y

Confirm pip2 is default pip
> pip --version
> pip install motioneye
> mkdir -p /etc/motioneye
> cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
> mkdir -p /var/lib/motioneye
> cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
> systemctl daemon-reload
> systemctl enable motioneye
> systemctl start motioneye

Was able to log into http://ip_address:8765 from Win10 to Pi3B+ no issue.
(updated Install Instructions testing date for RPiOS)

manno-xx, 2022-01-01:
Thanks for checking! Will follow steps above and report back asap.

starbasessd, 2022-01-01:
pi@rpi3-test:~ $ journalctl | grep -E "meyectl|motion"
Jan 01 15:10:10 rpi3-test systemd[1]: Started motionEye Server.
Jan 01 15:10:13 rpi3-test meyectl[367]:     INFO: hello! this is motionEye server 0.42.1
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: cleanup started
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: wsswitch started
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: tasks started
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: mjpg client garbage collector started
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: server started
- - - - - 
pi@rpi3-test:~ $ sudo systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2022-01-01 15:10:10 EST; 2min 42s ago
 Main PID: 367 (meyectl)
    Tasks: 5 (limit: 2059)
   CGroup: /system.slice/motioneye.service
           ├─367 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
           └─530 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

Jan 01 15:10:10 rpi3-test systemd[1]: Started motionEye Server.
Jan 01 15:10:13 rpi3-test meyectl[367]:     INFO: hello! this is motionEye server 0.42.1
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using defa
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: cleanup started
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: wsswitch started
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: tasks started
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: mjpg client garbage collector started
Jan 01 15:10:20 rpi3-test meyectl[367]:     INFO: server started

starbasessd, 2022-01-01:
BTW no other software installed, no system settings changed except the raspi-config 4 settings. No changes in motioneye but was able to access the WebGUI no issues, when I ran those 2 commands for comparison.

manno-xx, 2022-01-01:
Whoop whoop! Thanks. it worked. I also activated the camera in raspi-config while setting locales etc.

I did, again, get the error on the pillow install:
Collecting pillow (from motioneye)
  Downloading https://files.pythonhosted.org/packages/b3/d0/a20d8440b71adfbf133452d4f6e0fe80de2df7c2578c9b498fb812083383/Pillow-6.2.2.tar.gz (37.8MB)
    99% |############################### | 37.8MB 5.8MB/s eta 0:00:01Killed

but installing it using apt-get as mentioned in [https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) fixed that.

Side notes:
I do need to read up on a few things.
I get:
perl: warning: Falling back to a fallback locale ("en_US.UTF-8").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory

... a lot of the times when something is attempting accessing locales I guess. (I have it set to EN-US). Have seen such errors before but opportunistically decided to ignore

tbh, I'm not sure how to exactly distinguish between the pips. This is what I get though:
root@zero2wbirdcam:/home/pi# pip --version
pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)

Thanks again. Come daylight, I'll set up more with the camera.

Kind regards,
Manno

starbasessd, 2022-01-01:
I get a lot of those errors if I uncheck the en_GB.UTF-8 when setting locales. If I leave it checked, enable en_US.UTF-8, then select the en_US to be default in the next window, I lose most of the errors (not all, haven't figured out why, but it hasn't been critical for anything I do). You may have both python 2.7 and python 3.7.x installed. To see which is which, you run the following commands:
> python --version  \# gives default python
> pip --version  \# gives default pip
> python2 --version  \# gives current python2 version
> pip2 --version  \# gives the current pip2 version
> python3 --version  \# gives current python3 version
> pip3 --version  \# gives the current pip3 version

If python ever seems to be not responding as expected, it's usually because either the wrong version is default or the needed version isn't installed.  If you have python2 or pip2 as default, you can force the python3 by calling it explicitly:

> python3 pythonscript.py

and vice-versa...

manno-xx, 2022-01-01:
Thanks a bunch for the help!

closed 2022-01-01 by manno-xx