motioneye-project/motioneye #2261: get-pip.py ERROR - Install-on-Raspbian-Bullseye

wyskiki, 2022-01-02:
Hi everyone,

After 3 hours i'm getting crazy !!

I try to install motioneye (pi_buster_motion_4.4.0) on my Raspberry Pi 3 with Raspbian 11 (bullseye).
I followed the wiki [https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye](https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye) but i have an error on this command :

`python2 get-pip.py`

> pi@raspberrypi:~ $ sudo python2 get-pip.py
> ERROR:root:code for hash md5 was not found.
> Traceback (most recent call last):
>   File "/usr/local/lib/python2.7/hashlib.py", line 147, in <module>
>     globals()[__func_name] = __get_hash(__func_name)
>   File "/usr/local/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
>     raise ValueError('unsupported hash type ' + name)
> ValueError: unsupported hash type md5
> ERROR:root:code for hash sha1 was not found.
> Traceback (most recent call last):
>   File "/usr/local/lib/python2.7/hashlib.py", line 147, in <module>
>     globals()[__func_name] = __get_hash(__func_name)
>   File "/usr/local/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
>     raise ValueError('unsupported hash type ' + name)
> ValueError: unsupported hash type sha1
> ERROR:root:code for hash sha224 was not found.
> Traceback (most recent call last):
>   File "/usr/local/lib/python2.7/hashlib.py", line 147, in <module>
>     globals()[__func_name] = __get_hash(__func_name)
>   File "/usr/local/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
>     raise ValueError('unsupported hash type ' + name)
> ValueError: unsupported hash type sha224
> ERROR:root:code for hash sha256 was not found.
> Traceback (most recent call last):
>   File "/usr/local/lib/python2.7/hashlib.py", line 147, in <module>
>     globals()[__func_name] = __get_hash(__func_name)
>   File "/usr/local/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
>     raise ValueError('unsupported hash type ' + name)
> ValueError: unsupported hash type sha256
> ERROR:root:code for hash sha384 was not found.
> Traceback (most recent call last):
>   File "/usr/local/lib/python2.7/hashlib.py", line 147, in <module>
>     globals()[__func_name] = __get_hash(__func_name)
>   File "/usr/local/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
>     raise ValueError('unsupported hash type ' + name)
> ValueError: unsupported hash type sha384
> ERROR:root:code for hash sha512 was not found.
> Traceback (most recent call last):
>   File "/usr/local/lib/python2.7/hashlib.py", line 147, in <module>
>     globals()[__func_name] = __get_hash(__func_name)
>   File "/usr/local/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
>     raise ValueError('unsupported hash type ' + name)
> ValueError: unsupported hash type sha512
> Traceback (most recent call last):
>   File "get-pip.py", line 23974, in <module>
>     main()
>   File "get-pip.py", line 199, in main
>     bootstrap(tmpdir=tmpdir)
>   File "get-pip.py", line 83, in bootstrap
>     from pip._internal.commands.install import InstallCommand
>   File "/tmp/tmpiWvuTr/pip.zip/pip/_internal/commands/install.py", line 17, in <module>
>   File "/tmp/tmpiWvuTr/pip.zip/pip/_internal/cli/req_command.py", line 18, in <module>
>   File "/tmp/tmpiWvuTr/pip.zip/pip/_internal/index/collector.py", line 14, in <module>
>   File "/tmp/tmpiWvuTr/pip.zip/pip/_vendor/requests/__init__.py", line 43, in <module>
>   File "/tmp/tmpiWvuTr/pip.zip/pip/_vendor/urllib3/__init__.py", line 13, in <module>
>   File "/tmp/tmpiWvuTr/pip.zip/pip/_vendor/urllib3/connectionpool.py", line 11, in <module>
>   File "/tmp/tmpiWvuTr/pip.zip/pip/_vendor/urllib3/connection.py", line 15, in <module>
>   File "/tmp/tmpiWvuTr/pip.zip/pip/_vendor/urllib3/util/__init__.py", line 8, in <module>
>   File "/tmp/tmpiWvuTr/pip.zip/pip/_vendor/urllib3/util/ssl_.py", line 8, in <module>
> ImportError: cannot import name md5

I think i've tried everything... If you could help me... Thanks :)

starbasessd, 2022-01-02:
Why did you get "(pi_buster_motion_4.4.0)" if you are trying to do Bullseye?
Which version of Bullseye? 30 Oct 2021 or something else?
How did you install?
I want to try to reproduce as close as I can...

wyskiki, 2022-01-02:
I got the last release 4.4.0, just to be up to date.
Yes, it's the 30 Oct 2021 version.

First i followed the Wiki "Install for Raspbian", which is for Buster version (i didn't see that, sorry).
python-pip could not be installed, it was said "you can try with python3". So i installed python3-pip and followed the next steps. At the end, the motioneye status was "failed with result 'exit-code".

Then i tried the Wiki for Bullseye version and i got this MD5 error. And nothing on Google to fix this.

starbasessd, 2022-01-02:
https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye
motionEye is dependent on Python2.7 and pip2.
It may be best to start over again with a clean install of Bullseye and follow these instructions:
https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye
I am going through them now. If you tried using pip3 to install without any of the python2 or pip 2 support, it's not going to work, and I don't know how to clean up the bad installs, other than starting over.

starbasessd, 2022-01-02:
I'm at system update (42 files updating)... Should be up in about 30-45 minutes.

wyskiki, 2022-01-02:
Thank you for your help starbasessd !!! :)

Ok, i start again right now ! (I should have done that before posting :p)

wyskiki, 2022-01-02:
Error on 

`python2 get-pip.py`

python2: can't open file 'get-pip.py' : [Error 2] No such file or directory

starbasessd, 2022-01-02:
Didn't 
> curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py

download anything? I just ran that and it worked, as did
> python2 get-pip.py

starbasessd, 2022-01-02:
install 30 Oct 2021 official Bullseye lite
copy wpa_supplicant.conf and ssh files to /boot partition
install to Pi3B+
Run 
>sudo apt update && sudo apt upgrade -y

run 
> sudo raspi-config 

and set Locale, Timezone, password, and hostname
reboot

> sudo su
> apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y
> wget https://github.com/Motion-Project/motion/releases/download/release-4.3.2/pi_buster_motion_4.3.2-1_armhf.deb 
> dpkg -i pi_buster_motion_4.3.2-1_armhf.deb 
> systemctl stop motion
> systemctl disable motion 
> apt-get install python2 python-dev-is-python2 -y
> curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py
> python2 get-pip.py
> apt-get install libssl-dev libcurl4-openssl-dev libjpeg-dev zlib1g-dev -y
> pip install motioneye
> mkdir -p /etc/motioneye
> cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
> mkdir -p /var/lib/motioneye
> cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
> systemctl daemon-reload
> systemctl enable motioneye
> systemctl start motioneye

I was then able to connect to http://ip_address:8765 no issues

wyskiki, 2022-01-02:
Everything is working now. I'm sorry starbasessd ! My installation with python3 should be the origin of the problem...

Thank you for your time.

starbasessd, 2022-01-02:
What happened to your deleted message?
> Error on
> python2 get-pip.py
> python2: can't open file 'get-pip.py' : [Error 2] No such file or directory

Did you miss the curl line before it?

starbasessd, 2022-01-02:
And don't forget to close it if you are up and running...

wyskiki, 2022-01-02:
> What happened to your deleted message?
> 
> > Error on
> > python2 get-pip.py
> > python2: can't open file 'get-pip.py' : [Error 2] No such file or directory
> 
> Did you miss the curl line before it?

Yes, my mistake...

closed 2022-01-02 by wyskiki