motioneye-project/motioneye #2263: Can I install motioneye manually on a RPi zero 2 W?

rogierlommers, 2022-01-06:
In the wiki section, I found the excellent documentation which describes how to manually install motioneye on buster ([see here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)).

Now I'm wondering, is it possible to do that on a raspberry pi zero 2 W ? I know that for motioneyeos, there is someone who actually created a fork to make motioneyeos compatible with the zero 2 W. This makes me wondering if it's possible at all to manually install motion and motioneye on a zero 2 w.

starbasessd, 2022-01-06:
Yes. Please follow the appropriate set of instructions (Buster or Bullseye) and use motion 4.3.2.

gatozeko196, 2022-01-16:
https://github.com/ccrisan/motioneye/issues/2263#issuecomment-1006528894

starbasessd, 2022-01-16:
@gatozeko196 You linked back to this issue. Did you mean to do something else?

MichaIng, 2022-03-12:
The new `dev` branch has support for Python 3 and can hence be installed on Debian/Raspbian Bullseye. Depending on how you want to install `pip` and whether `motionEye` shall be installed as system-wide module or into a local directory, two options:
- https://github.com/motioneye-project/motioneye/pull/2326#issuecomment-1064709389
- https://github.com/cclauss/motioneye/blob/dev/.github/workflows/ubuntu_build.yml

JamBalaya56562, 2026-11-10:
Closing, as this was answered: yes, motionEye installs on a Pi Zero 2 W like on any Raspberry Pi OS, now with the venv steps in the README.

closed 2026-11-10 by JamBalaya56562