motioneye-project/motioneye #2264: Can not save Telegram Settings

djusHa, 2022-01-07:
Hi,

if i configure Telegram under "Motion Notifications" and click Apply, Settings are not saved.
Instead of this, "Run an End Command" ist activated with that CMD String that belongs to Telegram Settings.

[image]

[image]

starbasessd, 2022-01-15:
That option is not part of motionEye, it must be an add-on from another source. Please contact the author of the modification.

starbasessd, 2022-01-16:
@HawtDogFlvrWtr Can you see what, if anything is set up wrong here? I am unable to get this option to even appear, let alone test. It appears it should be available per https://github.com/ccrisan/motioneye/pull/1723

zagrim, 2022-01-16:
@djusHa did you try the Test-button?

I don't use Telegram and have never used this feature, and I wonder if there is a bug (say, if there is an error with the config, or contacting the Telegram API fails), or if it is your setup that has an issue. How did you install ME (Docker or standalone build directly from the code repository)? On which OS/distro? Have you checked the logs for errors?

djusHa, 2022-01-16:
Yes, test is succesfully, i can receive the Message on Telegram.
Only Issue is on save.

ME runs on Docker.

Build Image myself, with Files from ME Repo.

starbasessd, 2022-01-16:
Which volumes do you have shared in your docker run command? (-v folder_path:folder_path)

djusHa, 2022-01-16:
These:
```
volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./data/config:/etc/motioneye
      - ./data/videos:/var/lib/motioneye
```

starbasessd, 2022-01-16:
Can you change the ./data to the full path names and see if that helps?

djusHa, 2022-01-16:
I would, but have no working cam now.
My ESP32 Cam brocke yesterday...

starbasessd, 2022-01-16:
OK, let us know when you can test again.

djusHa, 2022-01-16:
ok

zagrim, 2022-01-16:
In the meantime, to me this looks like ME for some reason wouldn't be able to interpret the Telegram part of the camera config. I mean it is shown in the Run an end command field, which is really the place that ends up in the camera config file, but looking at the PR of the code changes for this feature, it should also be able to interpret that line so that the UI controls would show the settings (like for any other config params that are supported by ME). And looking at that code which should interpret the camera config, I'm wondering if your setup has something that is not matching the assumptions in the code.

When you click on Apply, is there any error shown in the UI? I would assume not, since you seem to have the UI updated anyway (the Run an end command field).
Could you show us the line from the camera config with the word "sendtelegram"? (if there's something that should be redacted please mask the sensitive parts but do it so that the number of parameters/words remains correct)

starbasessd, 2022-01-16:
@djusHa 
I am building a new docker image 
Host: Debian 11 Bullseye / 4GB / 50 GB / 4 cores on VBox VM fully updated
Following the instructions here: https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
> git clone -b dev https://github.com/ccrisan/motioneye.git
> cd motioneye
> docker build --build-arg VCS_REF=$(git rev-parse HEAD) \\ 
> --build-arg BUILD_DATE=$(date +"%Y-%m-%dT%H:%M:%SZ") \\ 
> -t ccrisan/motioneye:master-amd64 -f extra/Dockerfile .

Will run with:
>docker run --name="motioneye" \\
>    -p 8765:8765 \\
>    --hostname="motioneye" \\
>    -v /etc/localtime:/etc/localtime:ro \\
>    -v /etc/motioneye:/etc/motioneye \\
>    -v /var/lib/motioneye:/var/lib/motioneye \\
>    --restart="always" \\
>    --detach=true \\
>    ccrisan/motioneye:master-amd64

Let me know if you did anything different in your build process.

starbasessd, 2022-01-16:
Also if I read the manual and these instructions correctly, in the 
-v /etc/localtime:/etc/localtime:ro \
-v /etc/motioneye:/etc/motioneye \
-v /var/lib/motioneye:/var/lib/motioneye \

section, the first part is the internal to the container path, the second is the host path, so you might have the ./data paths backwards?

starbasessd, 2022-01-16:
OK, it's installed. I went to create an account to test with, and it demands my phone number. That isn't happening. I'm sorry, but I am going to leave any support up to others for this subject.

djusHa, 2022-01-16:
Thanks for your effort.

If your Docker Image is up an running, try to configure some cam and try to input ANY data under "Send a Telegram Notification", then click "Apply".

Do you have the same behaviour as me?

starbasessd, 2022-01-16:
I enabled still pics.
I enable Telegram
I add 12345678:AABBCC & 123456789 and 10 seconds
I do not get an "Apply" button at the top.
If I change something else (like video rotation) don't click apply, and try to add
12345678:AABBCC & 123456789 and 10 seconds
It still goes back to 'Off' for Telegram.
https://github.com/HawtDogFlvrWtr Is this a bug?

djusHa, 2022-01-17:
Yes, you have it and the Option "Run an End Command" is on with the Telegram CMD inside, right?

starbasessd, 2022-01-17:
Yes.

starbasessd, 2022-01-17:
@HawtDogFlvrWtr Tagging you for diagnostics

HawtDogFlvrWtr, 2022-01-17:
I'm out of town at the moment, but will look into it when I get back. Don't have the means to look into here here.

gatozeko196, 2022-02-19:
> Hi,
> 
> if i configure Telegram under "Motion Notifications" and click Apply, Settings are not saved.
> Instead of this, "Run an End Command" ist activated with that CMD String that belongs to Telegram Settings.
> 
> [image]
> 
> [image]
> 
>

MichaIng, 2022-04-12:
Solved in `dev`.

closed 2022-04-12 by MichaIng