motioneye-project/motioneye #2265: "no device" when starting wifi on RPi Zero 2 W on Jaspers image

derolly, 2022-01-07:
When trying to boot Jaspers Image for the Raspberry Pi Zero 2 W I get the following error:

` * Detecting disk device: /dev/mmcblk0
 * Checking data filesystem: done
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: none
 * Starting wpa_supplicant: no device
 * Configuring wireless network: no device
 * Configuring wired network: no device`

I get this error on 2 different boards with this image. Using the official Raspbian Image I do not get this error. I also tried this on 2 differently named wifi networks with the same result. When using a usb-to-network adapter it boots sucessfully. I checked then that the wpa_supplicant.conf was written correctly, and it was.

Any ideas? Maybe a missing driver or something?

starbasessd, 2022-01-07:
Sounds like a corrupted image. The whole point of @jawsper s image is because the 'official' motionEyeOS images don't have support for the wifi chip on the PiZero2W.

starbasessd, 2022-01-07:
Can you post your wpa_supplicant.conf file (blot out the SSID and password)
Make sure that the wpa_supplicant.conf doesn't end with .txt if created on a Win10 machine. 
Is it the same file that is copied to the /boot partition on the RPiOS image?

derolly, 2022-01-07:
The content of my wpa_supplicant.conf:

`country=DE
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
	scan_ssid=1
	ssid="MYSSID"
	psk="MYPASSWD"
}`

and the extension is just .conf not .txt

I downloaded the image again, and tried it with balenaEtcher and the Raspberry Pi Imager. No success.

Is it possible that I do not have the same wifi chip as you? How can I check that?

starbasessd, 2022-01-07:
You are sure you have the PiZero2W card? What does it have for model number on the back?
I haven't heard of any 'newer' models, yet.
You state the PiZero2Ws connect on wifi using RaspberryPiOS? Without the usb to ethernet adapter?
Also assuming the ` at the beginning of the first line and end of the last line are not in the actual file...

MStcz, 2022-01-15:
Hi Derolly, I am facing very same issue.
Here is the link to img I am trying: 

https://github.com/jawsper/motioneyeos/releases/download/20211113/motioneyeos-raspberrypi3.img.gz

And here what is containing wpa_supplicant.conf file. I have used Linux enters "LF".

country=CZ
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
ssid="ssid"
psk="psswd"
}

starbasessd, 2022-01-15:
For parsing issues, can you move the country=CZ to the 3rd line?
Can you add key_mgmt=WPA-PSK to the line after psk= and before }

MStcz, 2022-01-15:
Hi Starbasessd,

just tested with the same result.

Here is the config file:
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=CZ

network={
ssid="ssid"
psk="passwd"
key_mgmt=WPA-PSK
}

Console output:
- booting motionEyeOS unknown
* Detecting disk device: /dev/mncblko
* Creating data partition: done
* Formatting data partition: done
* Mounting data partition: done
* Copying data skeleton: done
* Mounting filesystems: done
* Mounting overlay filesystems: done
* Setting empty root password: done
* Copying version file: done
* Setting UTC timezone: done
* Loading kernel modules: done
* Setting hostname: done
* Configuring CPU frequency: done
* Starting syslogd: done
* Loading device-tree overlays: done
* Starting throttle watcher: done
* Starting eudeu: done
* Starting watchdog: done
* Starting rngd: done
* Preparing AP interfaces: none
* Starting wpa _supplicant: no device
* Configuring wireless network: no device
* Configuring wired network: no device
panic: rebooting (caused by network)
- shutting down motionEyeOS unknown
* Stopping motioneye: failed
* Stopping smbd: failed
* Stopping nbd: failed
* Stopping proftpd: failed
* Stopping sshd: failed
* Stopping crond: failed
* Stopping ntpd: failed
* Stopping network: failed

starbasessd, 2022-01-15:
If it were an official image, I'd be questioning:
/dev/mncblko
that would be /dev/mmcblk0 , but maybe @jawsper has an idea?

MStcz, 2022-01-15:
Sorry, /dev/mncblko is my fault. I took a picture of the screen and did OCR. I can see "0" (zero) on the picture but OCR wrote it as "o". Again sorry for this confusing.

jawsper, 2022-01-15:
Hi @MStcz, can you check if there's a file called `bcm2710-rpi-zero-2.dtb` in your boot partition? (the same as where you put the `wpa_supplicant.conf`)

MStcz, 2022-01-15:
Hi @jawsper, it is there. MD5 of the file is  E3C5B4E88C84C8351E8FD44C4FCD652B.

jawsper, 2022-01-15:
Hmm that matches the one I have. Have you tried running the official Raspberry Pi OS on this specific Pi, to rule out a hardware issue?

derolly, 2022-01-15:
On my two Pi zero 2 w the official Raspbian Buster runs perfect. Thats what I'm using motioneye on right now. But your image still does not work with both zero 2 w.

MStcz, 2022-01-15:
This image is working with the same wpa_supplicant.conf config file.
https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2021-11-08/2021-10-30-raspios-bullseye-armhf.zip

dvandijk, 2022-01-19:
I'm having the same issue with zero results in changing the wpa_supplicant.conf (I've tried them all). I have another Pi Zero2 which is working fine. Using that image doesn't work either.

The only difference between the Pi Zero2's are the version and chip serial.

Working: version 1.5 with chip serial 210340G
Not working: version 2.11 with chip serial 210340E

starbasessd, 2022-01-19:
@dvandijk 
Interesting. Can you post a pic of both PiZero2W backs (where it shows the version number)? I have been told in the RPi forums that there aren't 2 versions, but that would explain why there are more and more issues...

jawsper, 2022-01-19:
Ah that is very interesting! Seems there could be a new chip revision, possibly requires another rpi-firmware update.

dvandijk, 2022-01-19:
Sure, hope this helps

[image]

jawsper, 2022-01-19:
Interesting, the main CPU does seem to be identical, but perhaps they had to update the Wifi chip. I'll build an image with the latest firmware later tonight.

starbasessd, 2022-01-19:
@dvandijk Any chance you could get the identifying info off the silver WiFi chips? Please?

jawsper, 2022-01-19:
@dvandijk Can you try this new version? https://github.com/jawsper/motioneyeos/releases/tag/20220119-dev
I updated rpi-firmware (not sure if needed) and rpi-wifi-firmware (this may fix your issue).

jawsper, 2022-01-19:
(I'm not near my Pi so this image is not tested by me)

derolly, 2022-01-19:
I tried the image and can confirm that it works on my pi! Thank you @jawsper 👍

dvandijk, 2022-01-19:
Can also confirm that it works! Thanks indeed @jawsper

MStcz, 2022-01-19:
Hi @jawsper, great job, thank you for the solution.

john3parker, 2022-01-30:
HI @jawsper, thank you for the build. It works for my PiZero2W when other builds would not. However, applying config changes to the camera with you build yields this error for my Remote MotionEye Camera.

`2022-01-30 03:02:17: [motioneye]    ERROR: 'privacy_mask'
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1756, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 297, in post
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 537, in set_config
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 376, in set_camera_config
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 834, in motion_camera_ui_to_dict
KeyError: 'privacy_mask'
2022-01-30 03:02:17: [motioneye]    ERROR: 500 POST /config/1/set/?_username=admin&_admin=true&_signature=adb1e6260cf73e1d5c822e8f0aba57e1f72fa5b9 (192.168.1.31) 15.75ms
`

starbasessd, 2022-01-30:
I do notice an inconsistent use of "/" leading (missing in the first 2 lines), maybe a bad edit?
Are both motionEyes running on PiZero2Ws? Or which is which?

john3parker, 2022-01-31:
> I do notice an inconsistent use of "/" leading (missing in the first 2 lines), maybe a bad edit? Are both motionEyes running on PiZero2Ws? Or which is which?

I doubled check and was able to reproduce the missing "/" - its not a bad edit. I am running PiZero2W. 

[image]

starbasessd, 2022-01-31:
@jawsper thoughts?

john3parker, 2022-02-01:
@jawsper Looks like this is a missing `privacy_mask` setting in config.py for _set_default_motion_camera. This sets the default camera_config (line 567) and sent to motion_camera_ui_to_dict (line 570) but privacy_mask doesn't have a default value. It's referenced on line 834 and the error is thrown. May want to solve for this by setting an initial value for privacy_mask = False and the checking for the True/False instead whether the value actually exists. 
I'm not a Python developer and not really sure how the packaging/builds work or I'd give that a try myself.

john3parker, 2022-02-05:
More debugging - I'm running a central server with MotionEye web frontend server and an attached MotionEyeOS camera. Configuring the camera directly from the camera IP it works fine. The request payload clearly has entries privacy_mask and privacy_mask_lines. 
```
{
  "enabled": true,
  "name": "Camera11",
  "proto": "mmal",
  "auto_brightness": false,
  "rotation": "0",
  "framerate": "2",
  "privacy_mask": false,
  "privacy_mask_lines": [],
  "extra_options": [],
  "storage_device": "custom-path",
```

When I try to configure the camera from the MotionEye frontend server it throws the privacy_mask error. Here you can see the request payload is missing the privacy_mask attribute.
```
{
    "enabled": true,
    "name": "Camera11",
    "proto": "motioneye",
    "auto_brightness": false,
    "rotation": "0",
    "framerate": "2",
    "extra_options": [],
    "storage_device": "custom-path",
```

john3parker, 2022-02-05:
I'm running the local MotionEye server from a [docker container](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker). It's version is 0.42. There seems to be a mismatch in versions between these builds. Can you update the docker image with the latest version?
```
#      
# meyectl -v
motionEye 0.42
# 

```
The camera is running version 0.42.1 
```
[root@meye-2854046f ~]# meyectl -v
motionEye 0.42.1
[root@meye-2854046f ~]# 
```

starbasessd, 2022-02-05:
How are you logging into the docker container to run meyectl -v? 
What does the WebGUI report for the docker container as to version for motion, motionEye, and OS?

starbasessd, 2022-02-05:
Nevermind. 
Confirmed 0.42.0 for the docker image. Along with motion 4.2.2 and OS Ubuntu 18.10.
Checking the Install in Docker [Build Instructions](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker#build-instructions), will report shortly.

starbasessd, 2022-02-05:
OK, Build Docker gives motionEye 0.42.1, but motion 4.1.1, and OS is Debian 10. Will look at it further...

starbasessd, 2022-02-05:
I do not have enough time to figure out how to add the binary from motion-project, rather than the default motion from the Debian 10 repo used by the docker build... Anyone want to make suggestions, I'll be happy to re-write the Install in Docker instructions to update them. It is much more viable for the 'average' user (read: script kiddie like me) to build the latest docker than build a motionEyeOS image.

jawsper, 2022-02-05:
So the version of motioneye on motioneyeos doesn't seem to be actually 0.42.1, there just hasn't been a bump to the version numbers since then, but @ccrisan did update motioneye since then, and that privacy mask feature was added after the 0.42.1 release. The actual version is [1b04e8968c2a127b23341575ab80e9892893b576](https://github.com/ccrisan/motioneye/tree/1b04e8968c2a127b23341575ab80e9892893b576)

starbasessd, 2022-02-05:
dev20201026 does have the privacy mask option & 0.42.1, with motion 4.3.1. Privacy Mask in motion first appeared 4.1

starbasessd, 2022-02-05:
Working on adding motion 4.3.2-1 to amd64 Dockerfile build. Suggestions on how to add [1b04e8968c2a127b23341575ab80e9892893b576](https://github.com/ccrisan/motioneye/tree/1b04e8968c2a127b23341575ab80e9892893b576)?

starbasessd, 2022-02-06:
If there is any interest, I can try for updating to Bullseye, or including RPi specific changes, too... I am not anywhere near a Dev or Programmer, but want to extend usability as long as possible, and the docker container may be the way to go for motionEye.

starbasessd, 2022-02-06:
OK, have new Dockerfile working, gives Debian 10 (Buster) OS, latest motionEye (with Privacy Mask) and motion 4.3.2-1 
>    attached as Dockerfile.zip 
>    unzip  and place in motioneye/extra folder replacing existing Dockerfile

Command to build:

>     docker build --build-arg VCS_REF=$(git rev-parse HEAD) \
>         --build-arg BUILD_DATE=$(date +"%Y-%m-%dT%H:%M:%SZ") \
>         -t ccrisan/motioneye:master-amd64 -f extra/Dockerfile .
> 

command to run:
>    docker run --name="motioneye" -p 8765:8765 --hostname="motioneye" \
>        -v /etc/localtime:/etc/localtime:ro \
>        -v /etc/motioneye:/etc/motioneye \
>        -v /var/lib/motioneye:/var/lib/motioneye \
>        --restart="always" \
>        --detach=true ccrisan/motioneye:master-amd64
 
If someone would test it, and let me know it's working as expected, I will then add to Install in Docker documentation. 

[Dockerfile.zip](https://github.com/ccrisan/motioneye/files/8009217/Dockerfile.zip)

john3parker, 2022-02-06:
@starbasessd i was able to build with the new Dockerfile and deploy the container. Everything appears to be working perfectly for me. MotionEye 0.42.1 (with Privacy Mask) and motion 4.3.2. My host operating system is Ubuntu 20.04. I was able to add and update cameras from MotionEye server. Looks good to me! Thank you for doing all the heavy lifting!

fils02, 2022-08-24:
Hi @jawsper, I found your build for motioneyeos rpi0 2 w, and I tried to run but it seems not connecting to wifi even if I tried a lot of times with different wpa_supplicant files. Then I found this topic about the two version of chip and realized that mine is a third one version!
I was wondering if you could add a new version for this. 

 chip serial RP3A0-AU
                    2104
                 201221A

Thank you very much in advance

Slipery, 2023-10-08:
Looks like this problem has raised its head again.  Pi Zero 2W marked as version 4.6
on boot up I get 
Starting wpa _supplicant: no device
Configuring wireless network: no device
Configuring wired network: no device
panic: rebooting (caused by network)

C4pric, 2023-12-25:
hi @jawsper, I have a Raspberry pi zero 2w and I have the same problems as many, my wpa_supplicant file does not detect any service. I would like to know if it was possible to update motioneyes for rpi2w. My raspberry 2w is part of version 4.2.

Rileyjarvis, 2024-12-27:
@jawsper hi there im having issues were wireless is not an option in the boot up it goes wired network no devices rebooting caused by network my raspi zero 2w is -
RP3A0-AU
2413
240206B
and is v4.6 
[image]
[image]
[image]

RedKeyEng, 2025-01-11:
Joined GitHub solely to say that after 2 days trying everything, modifying wpa_supplicant files to various formats, I've discovered that Raspberry Pi Imager doesn't work with this image. Use only Balena Etcher when flashing the image. Worked perfectly for me.

Rileyjarvis, 2025-01-13:
> Joined GitHub solely to say that after 2 days trying everything, modifying wpa_supplicant files to various formats, I've discovered that Raspberry Pi Imager doesn't work with this image. Use only Balena Etcher when flashing the image. Worked perfectly for me.

Still having issues even with a fresh image of jasper's version of the os just incase it got messed up and its still doing it ive used the flasher you said to use at it had no difference it still panics and reboots due to network (no device found)
Im wondering if theres an update for the chip i haven't found yet or if ive cooked it with octopi which ran perfectly fine while i needed it pervious.

jawsper, 2025-01-13:
For any of these kind of issues, can you also make sure to test with a regular Raspbian image, just to exclude any hardware issues?

Rileyjarvis, 2025-01-13:
> For any of these kind of issues, can you also make sure to test with a regular Raspbian/Raspberry Pi OS image, just to exclude any hardware issues?

Yea sure ill try that now any particular version i should try?

Rileyjarvis, 2025-01-13:
> For any of these kind of issues, can you also make sure to test with a regular Raspbian/Raspberry Pi OS image, just to exclude any hardware issues?

Wifi works and everything on raspberry pi os

MichaIng, 2025-10-04:
A known issue since the motionEyeOS image uses a too old kernel.

Let me move this issue to the respective repository.