motioneye-project/motioneye #2266: Writing video streams to disk fails, codec error?

rogierlommers, 2022-01-08:
I want to enable the writing of video streams to disk.
For some reason it fails, in the logs I see below messages.
I followed these instructions: https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye (from @starbasessd )
Did I miss something? Do I need to explicitly install a codec?

```
[1:ml1:Vogelhuis] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.
[1:ml1:Vogelhuis] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Encoder not found
[1:ml1:Vogelhuis] [ERR] [ENC] ffmpeg_open: Failed to allocate codec!
[1:ml1:Vogelhuis] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.

```

starbasessd, 2022-01-08:
Are you  using any "Extra Motion Options" or other modifications? I use 
[image]
and don't have issues

starbasessd, 2022-01-08:
You could try Movie Passthrough = ON (but check the tooltip at the end of that line to see the limitations)(just point your mouse at the ? don't click)

rogierlommers, 2022-01-11:
Passthrough also didn't work. Now I'm using `MPEG4` and it's working. However this is not a "native" MacOS readable format, so I need to manually convert the files :(.

starbasessd, 2022-01-11:
You should be able to use VLC for Mac to view it 'natively'.

rogierlommers, 2022-01-11:
> You should be able to use VLC for Mac to view it 'natively'.

Yeah true, but I would like to drag them on the Kids' iPads :).
Requires ffpmeg now :)

`find /media/motioneye/Drive/vogelhuis-media -type f -iname "*.avi" -exec bash -c 'ffmpeg -i $0 -c:v copy -c:a copy -y ${0/avi/mp4} && rm ${0}' {} \;`

closed 2022-01-11 by rogierlommers