motioneye-project/motioneye #2267: Video artefacts in web view

hanscomps, 2022-01-08:
Archlinux
motioneye 0.42-1 (0.42, 0.41)
motion 4.4.0-1 (4.3.2-1)
python 2.7.18

After a long overdue update I now get weird color artefacts in the motioneye web-view, camera video previews.  The recorded clips (on motion) are still fine; also checked that the camera rtsp feeds are working fine in VLC.
I have downgraded both motioneye and motion to previous version with no success.
Screenshot:  https://drive.google.com/file/d/1HuoFYGjEUu-gEZGu2VPFQlCqspscDG_C/view?usp=sharing

starbasessd, 2022-01-08:
Arch 5.15.11-arch2-1 on 4Cores / 4GB / 48GB
Latest system updates
Followed wiki install
Added first camera
Got the same issue.
Looked like a previous issue where there was too many characters in Text Overlay,
but that solution didn't work here. 
I believe it's related to motion 4.4.0, but I can't figure out how to downgrade to 4.3.2 or 4.2.2
Suggestions?

hanscomps, 2022-01-09:
I did downgrade to motion 4.3.2 (only other version I ever had installed) without success.
Do not think it is motioneye (not much changed in the last 2 years) or motion (recordings are fine).   Could be a python2 library, trying to figure out now what code is used to display the frontend videos.

starbasessd, 2022-01-09:
I honestly don't know. Arch is somewhat like Gentoo, where you can't really go back to a certain starting point, and come forward from there, like you can with the Debian or other flavors derivatives, as it is a 'rolling' image. I am cloning the image from my last successful build of arch to see if it is broken there, too, without doing a pacman -Syu update...

starbasessd, 2022-01-09:
I am able to get motionEye running in Arch, but had to use a side-door:
su -
pacman -Syu
pacman -S docker docker-compose
usermod -aG docker [your user]
systemctl enable docker
reboot
log back in
then use the docker start command here:
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
This bypasses any incompatibilities between the way software is built/compiled in Arch vs what is needed for motionEye/motion/ffmpeg.
I added 5 cameras with the following command:
> docker run --name="motioneye" \
>    -p 8765:8765  -p 8081:8081 -p 8082:8082 -p 8083:8083 -p 8084:8084 -p 8085:8085 \
>    --hostname="motioneye" \
>    -v /etc/localtime:/etc/localtime:ro \
>    -v /etc/motioneye:/etc/motioneye \
>    -v /var/lib/motioneye:/var/lib/motioneye \
>    --restart="always" \
>    --detach=true \
>    ccrisan/motioneye:master-amd64

And connected to http://arch_ip:8765 and added the 5 cameras and had clear video. There may be better ways to start this, but this is what I tested and it works.

hanscomps, 2022-01-10:
Can you please list the versions of the packages; python2, motioneye, motion, ffmpeg.
Also the output from >pip2 list

starbasessd, 2022-01-10:
In a docker container or from where?

black-roland, 2022-01-10:
The same issue happened to me after updating a system. I'm using Fedora 34 and motionEye is installed in a venv. `pip install --force-reinstall --no-cache-dir motioneye` didn't help.

black-roland, 2022-01-10:
In my case, artifacts started after I updated these packages:

<details>
<summary>dnf.rpm.log</summary>

```
2022-01-10T22:13:01+0300 SUBDEBUG Upgraded: libsmbclient-2:4.14.9-0.fc34.x86_64
2022-01-10T22:13:01+0300 SUBDEBUG Upgraded: samba-client-libs-2:4.14.9-0.fc34.x86_64
2022-01-10T22:13:01+0300 SUBDEBUG Upgraded: samba-common-libs-2:4.14.9-0.fc34.x86_64
2022-01-10T22:13:01+0300 SUBDEBUG Upgraded: libwbclient-2:4.14.9-0.fc34.x86_64
2022-01-10T22:13:01+0300 SUBDEBUG Upgraded: bind-utils-32:9.16.21-1.fc34.x86_64
2022-01-10T22:13:01+0300 SUBDEBUG Upgraded: alsa-utils-1.2.5.1-1.fc34.x86_64
2022-01-10T22:13:01+0300 SUBDEBUG Upgraded: python3-devel-3.9.7-1.fc34.x86_64
2022-01-10T22:13:01+0300 SUBDEBUG Upgraded: pango-devel-1.48.10-1.fc34.x86_64
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: python3-certbot-dns-cloudflare-1.20.0-1.fc34.noarch
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: python3-rpm-macros-3.9-38.fc34.noarch
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: systemtap-4.5-1.fc34.x86_64
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: libxcrypt-devel-4.4.26-2.fc34.x86_64
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: libsepol-devel-3.2-1.fc34.x86_64
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: fedora-release-34-37.noarch
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: fedora-release-common-34-37.noarch
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: python-rpm-macros-3.9-38.fc34.noarch
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: alsa-ucm-1.2.5.1-2.fc34.noarch
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: python-unversioned-command-3.9.7-1.fc34.noarch
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: perl-Mozilla-CA-20200520-4.fc34.noarch
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: certbot-1.20.0-1.fc34.noarch
2022-01-10T22:13:02+0300 SUBDEBUG Upgraded: systemtap-client-4.5-1.fc34.x86_64
2022-01-10T22:13:03+0300 SUBDEBUG Upgraded: systemtap-devel-4.5-1.fc34.x86_64
2022-01-10T22:13:03+0300 SUBDEBUG Upgraded: elfutils-devel-0.185-2.fc34.x86_64
2022-01-10T22:13:03+0300 SUBDEBUG Upgraded: python3-certbot-1.20.0-1.fc34.noarch
2022-01-10T22:13:03+0300 SUBDEBUG Upgraded: systemtap-runtime-4.5-1.fc34.x86_64
2022-01-10T22:13:03+0300 SUBDEBUG Upgraded: nss-3.71.0-1.fc34.x86_64
2022-01-10T22:13:03+0300 SUBDEBUG Upgraded: nss-softokn-3.71.0-1.fc34.x86_64
2022-01-10T22:13:03+0300 SUBDEBUG Upgraded: nss-sysinit-3.71.0-1.fc34.x86_64
2022-01-10T22:13:03+0300 SUBDEBUG Upgraded: libaom-3.1.1-1.fc34.x86_64
2022-01-10T22:13:03+0300 SUBDEBUG Upgraded: elfutils-libelf-devel-0.185-2.fc34.x86_64
2022-01-10T22:13:03+0300 SUBDEBUG Upgraded: nss-softokn-freebl-3.71.0-1.fc34.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: nss-util-3.71.0-1.fc34.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: grub2-tools-extra-1:2.06-6.fc34b.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: cups-client-1:2.3.3op2-7.fc34.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: python3-acme-1.20.0-1.fc34.noarch
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: elfutils-debuginfod-client-devel-0.185-2.fc34.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: elfutils-libs-0.185-2.fc34.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: elfutils-debuginfod-client-0.185-2.fc34.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: grub2-tools-minimal-1:2.06-6.fc34b.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: libjxl-0.5-1.fc34.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: bind-libs-32:9.16.21-1.fc34.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: python3-gammu-3.2.3-1.fc34.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: python3-tkinter-3.9.7-1.fc34.x86_64
2022-01-10T22:13:04+0300 SUBDEBUG Upgraded: python3-libs-3.9.7-1.fc34.x86_64
2022-01-10T22:13:05+0300 SUBDEBUG Upgraded: python3-3.9.7-1.fc34.x86_64
2022-01-10T22:13:05+0300 SUBDEBUG Upgraded: mosquitto-2.0.12-1.fc34.x86_64
2022-01-10T22:13:05+0300 SUBDEBUG Upgraded: shadow-utils-2:4.8.1-9.fc34.x86_64
2022-01-10T22:13:05+0300 SUBDEBUG Upgraded: libxcrypt-compat-4.4.26-2.fc34.x86_64
2022-01-10T22:13:05+0300 SUBDEBUG Upgraded: grub2-tools-efi-1:2.06-6.fc34b.x86_64
2022-01-10T22:13:05+0300 SUBDEBUG Upgraded: grub2-tools-1:2.06-6.fc34b.x86_64
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: glib-networking-2.68.2-1.fc34.x86_64
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: ca-certificates-2021.2.50-1.0.fc34.noarch
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: grub2-common-1:2.06-6.fc34b.noarch
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: tzdata-2021c-1.fc34.noarch
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: bind-license-32:9.16.21-1.fc34.noarch
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: elfutils-default-yama-scope-0.185-2.fc34.noarch
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: python-srpm-macros-3.9-38.fc34.noarch
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: fedora-release-identity-basic-34-37.noarch
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: samba-common-2:4.14.9-0.fc34.noarch
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: vim-filesystem-2:8.2.3512-1.fc34.noarch
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: sysprof-capture-devel-3.40.1-1.fc34.x86_64
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: rust-srpm-macros-18-1.fc34.noarch
2022-01-10T22:13:06+0300 SUBDEBUG Upgraded: kernel-headers-5.14.9-200.fc34.x86_64
2022-01-10T22:13:07+0300 SUBDEBUG Upgraded: hwdata-0.352-1.fc34.noarch
[... 6 lines cut ...]
2022-01-10T22:13:07+0300 SUBDEBUG Upgraded: alsa-lib-1.2.5.1-2.fc34.x86_64
2022-01-10T22:13:07+0300 SUBDEBUG Upgraded: libsepol-3.2-1.fc34.x86_64
2022-01-10T22:13:07+0300 SUBDEBUG Upgraded: pango-1.48.10-1.fc34.x86_64
[... cut ...]
2022-01-10T22:13:07+0300 SUBDEBUG Upgraded: libdrm-2.4.107-1.fc34.x86_64
2022-01-10T22:13:08+0300 SUBDEBUG Upgraded: libdav1d-0.9.1-1.fc34.x86_64
2022-01-10T22:13:08+0300 SUBDEBUG Upgraded: less-581.2-1.fc34.x86_64
2022-01-10T22:13:08+0300 SUBDEBUG Upgraded: jsoncpp-1.9.4-3.fc34.x86_64
2022-01-10T22:13:08+0300 SUBDEBUG Upgraded: bluez-libs-5.62-1.fc34.x86_64
2022-01-10T22:13:08+0300 SUBDEBUG Upgraded: bluez-5.62-1.fc34.x86_64
2022-01-10T22:13:08+0300 SUBDEBUG Upgraded: SDL2-2.0.16-3.fc34.x86_64
```

</details>

I don't see anything suspicious :thinking:

hanscomps, 2022-01-11:
> In a docker container or from where?
In the container yes; or just the versions of motioneye and motion that it uses.

hanscomps, 2022-01-11:
Did some investigation last night.  Web standards and streaming is not my area of expertise, but I can read python and C code.

Seems that at some point there was a small change in how the jpeg streaming or handling was done; specifically "Transfer-Encoding: chunked".   When I grab a "raw" stream from motion (4.4.0, 4.3.2, 4.3.1) I can see the same "artefacts" in the jpeg binary;  these are caused by the chunk "/r/n<chunk length in hex>/r/n" headers.  Removing all these header bytes corrects the jpeg.
It looks like the tornadoweb libs used by motioneye also just grabs the "raw" jpeg stream and does not handle the chuck headers?  These motion streams are handled correctly in browsers, so assuming it is a motioneye (support lib) problem. 

I will try and figure out what underlying library is responsible for this; but at this point I guess the developers would be best to point to the correct issue.

starbasessd, 2022-01-11:
I know motion is very sensitive to video size, dimensions need to be evenly divisible by 8 and 16, otherwise you have serious issues. You may want to see if this is part of your issue.

starbasessd, 2022-01-11:
> > In a docker container or from where?
> > In the container yes; or just the versions of motioneye and motion that it uses.

Interesting. as I do not have any issues with the versions in the official docker container
which uses 
motionEye Version | 0.42
Motion Version | 4.2.2
OS Version | Ubuntu 18.10
and the same 6 cameras I use for my 'production' system, at 640x480, 800x600, and 1280x720 for the cameras.

dodysaway, 2022-01-12:
It happens in Fedora 35 also with motion-4.4.0, motioneye 0.42.1 (0.42.0 too), ffmpeg 4.4.1.

starbasessd, 2022-01-12:
What resolutions are you using for your camera(s)?

dodysaway, 2022-01-12:
> What resolutions are you using for your camera(s)?

Initial: 1920 x 1080
Now: 800 x 600

starbasessd, 2022-01-12:
On my production server I have 
>ffmpeg --version
>ffmpeg version 4.1.8-0+deb10u1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 8 (Debian 8.3.0-6)

running on
>sudo lsb_release -a
>No LSB modules are available.
>Distributor ID: Debian
>Description:    Debian GNU/Linux 10 (buster)
>Release:        10
>Codename:       buster

PiZeroW running motionEyeOS dev20201026:
>ffmpeg version 4.3.1 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)

dodysaway, 2022-01-12:
I've just found the source of the issue:
libmicrohttpd-0.9.75

If I downgrade the library to: libmicrohttpd-0.9.73 the problem is solved.

starbasessd, 2022-01-12:
Interesting, RaspiOS loads a version directly, but the other Debian derivatives don't. I wonder what loads it as a dependency... Might have to do research if we start getting more issues from other users.

black-roland, 2022-01-12:
> If I downgrade the library to: libmicrohttpd-0.9.73 the problem is solved.

Worked for me too. Thanks! In my case, I downgraded `0.9.75-2` to `0.9.72-2` on Fedora 34.

black-roland, 2022-01-12:
> Interesting, RaspiOS loads a version directly, but the other Debian derivatives don't. I wonder what loads it as a dependency... Might have to do research if we start getting more issues from other users.

Motion itself uses `libmicrohttpd` if I understood you correctly: https://github.com/Motion-Project/motion/blob/34091a770c679da79f5be58f392020a8514521ee/configure.ac#L62

starbasessd, 2022-01-12:
My Debian Buster (with motion 4.3.2, motionEye 0.42.1) has
>sudo dpkg-query -l | grep libmicrohttpd
>ii  libmicrohttpd12                   0.9.62-1                      amd64        library embedding HTTP server functionality
and doesn't want to upgrade it.
Clean fully updated Debian 11 Bullseye:
sudo apt install motion
gives motion 4.3.2 and libmicrohttpd12 0.9.72-2
and doesn't try to update either.
Must be a RH issue or how are you upgrading?

hanscomps, 2022-01-12:
Downgrading libmicrohttpd to 0.9.73 works for me also.
I can confirm with this lib version motion does not use "Transfer-Encoding: chunked" when serving the mjpeg streams.
I assume because the tornado-web package dropped support for Python 2.7 going to version 6.0; support for receiving "chunks" will not be possible until the port to Python 3 is complete.
For now

starbasessd, 2022-01-12:
in Debian:
> wget https://github.com/Motion-Project/motion/releases/download/release-4.4.0/bullseye_motion_4.4.0-1_amd64.deb
> sudo dpkg --install bullseye_motion_4.4.0-1_amd64.deb
> sudo apt-get install --fix-broken    #to grab the missing dependencies
> sudo motion --version
> motion Version 4.4.0
> sudo dpkg-query -l | grep libmicrohttpd
> ii  libmicrohttpd12:amd64          0.9.72-2

Will check other OSs as I have time, but don't know where you picked up 
the bad libmicrohttp...

hanscomps, 2022-01-12:
> Downgrading libmicrohttpd to 0.9.73 works for me also. I can confirm with this lib version motion does not use "Transfer-Encoding: chunked" when serving the mjpeg streams. I assume because the tornado-web package dropped support for Python 2.7 going to version 6.0; support for receiving "chunks" will not be possible until the port of motioneye to Python 3 is complete. Alternatively can try and get the motion project to add a feature to disable chunk encoding on streams.

Going through the tornadoweb releases it looks like "Transfer-Encoding: chunked" was added at version 4.0.
So unless the function broke on the way to version 5.1.1, I assume it is possible to still fix it within the current supported packages?

starbasessd, 2022-01-14:
> Motion itself uses `libmicrohttpd` if I understood you correctly: https://github.com/Motion-Project/motion/blob/34091a770c679da79f5be58f392020a8514521ee/configure.ac#L62

@black-roland I went to the motion-releases page, grabbed https://github.com/Motion-Project/motion/releases/download/release-4.4.0/bullseye_motion_4.4.0-1_amd64.deb and installed. 
the internal dependency libmicrohttp12 it installed was -72. Where are you picking up later versions?

starbasessd, 2022-01-14:
Interesting. I am not familiar enough with either Fedora (haven't used in years) or Arch (just started trying to figure it out). Is there an easy way to downgrade the package in either system? I'd like to include instructions in the Wikis for both installs, if possible.

starbasessd, 2022-01-14:
Thank you! 
I will try to get the install Wiki for Fedora updated soon. What is the current version, so I may note it? May I tag you to check it for me when done?

bkiziuk, 2022-02-02:
I also have this issue on Debian testing.
For downgrading libmicrohttpd12 to the version pointed above:
1) add snapshot repository to `/etc/apt/sources.list`
```
deb http://snapshot.debian.org/archive/debian/20210816T091510Z/ testing main
```

2) refresh ignoring the InRelease expiration
```
sudo apt-get -o Acquire::Check-Valid-Until=false update
```

3) rollback the `libmicrohttpd12` package:
```
sudo apt install libmicrohttpd12=0.9.72-2
```

4) mark the package to be on hold to prevent further upgrades to it
```
sudo apt-mark hold libmicrohttpd12
```

alpytron, 2022-03-31:
Hi,
So is there any update on this? 
Downgrading libmicrohttpd12 does not seem to be a "solution" right?

alpytron, 2022-04-02:
Nope, I still have the issue with Python 3 version too:
motioneye-0.43.0 
[image]
The resolution is only 640x480.
I can confirm that the mjpeg stream from motion is OK, so the artifact/corruption comes from motioneye.

MichaIng, 2022-04-02:
When downgrading `libmicrohttpd12` solves it, then it is obvious.

Read further through the comments above, so it seems like we need to set an option `Transfer-Encoding: chunked` which was set by default on older `libmicrohttpd12`/`tornadoweb` versions. I'll have a closer look at this tomorrow.

jmichault, 2022-04-24:
A solution could be to send HTTP/1.0 headers in mjpgclient.py instead of HTTP/1.1.
This should force libmicrohttpd12 to not use chunks.

MichaIng, 2022-04-28:
Issue worked around in `dev`.