motioneye-project/motioneye #2268: ghosting when uploaded to google drive

m1eqb, 2022-01-10:
motioneye 20200606 
raspberry pi 3b
using 2x iegeek ip cams 
rtsp:// ....ip of cam.....79:554/11 camera 1 
rtsp:// ....ip of cam....173:554/11 camera 2
was working well with 1 camera now i have added a second camera i keep getting ghosting on the uploaded google drive files 
im i over loading the pi

https://user-images.githubusercontent.com/22138001/148843367-377e00fd-141f-49eb-b386-310538a8cb4f.mp4

many thanks for your help 
gerry

starbasessd, 2022-01-10:
This is the motionEye github, you need the motionEyeOS github issues:
https://github.com/ccrisan/motioneyeos/issues
Suggestions:
Lower resolution
Use motionEyeOS dev20201026
Q: Does it happen on both cameras?
Q: Does it happen if the movies are stored locally?
Q: What movie format are you using for storage?

starbasessd, 2022-01-10:
Are the cameras wireless or wired? If wireless, it could also be a bandwidth issue. Especially if a single camera works well. Is the Pi3 connecting to the network wirelessly, too? Or is it on Ethernet? If wireless, then you have 2 inbound video streams, and outbound stream(s) to the GDrive.

m1eqb, 2022-01-10:
yes they are all wireless so it looks like im over cooking the pi 
will have to look for a different way of runing the cams 
thanks for your help

starbasessd, 2022-01-10:
Also check your power & thermals. SSH or console and run the command:
vcgencmd get_throttled
If it reports anything but 0x0 post it here.

m1eqb, 2022-01-10:
may try adding second pi so each camera has it own unit

starbasessd, 2022-01-10:
Won't affect bandwidth. Would help CPU.

closed 2022-01-13 by starbasessd