motioneye-project/motioneye #2269: Question: Log4j Vulnerability

lozziboy, 2022-01-12:
On December 9, 2021 a vulnerability impacting versions 2.0-beta9 to 2.14.1 of the Apache Log4j 2 utility was announced on its GitHub page. With this ticket I would like to ask if motioneye uses any Log4j directly or indirectly.
This is just to investigate about possible security incidents.

zagrim, 2022-01-12:
ME is not using Log4j at least directly, but any indirect use is way harder to find out. There are ways to use Log4j from Python, but a quick search in the project files with "log4j" didn't find anything. Neither did I find any matches under Python dist-packages directory (where the dependencies are downloaded). Personally I'd go with assuming ME or it's dependencies (at least the standard ones mentioned in the installation instructions) do not have anything to do with Log4j, but that's only my assumption given the facts I have.

starbasessd, 2022-01-12:
As @zagrim noted, the log4j is not used in motionEye. MotionEye's web interface doesn't use Apache, it uses python2 (it's own can of worms). I would be concerned if it was running behind nginx or another apache based web redirector or encryption server. or anything running Java/Javascript.

MichaIng, 2022-03-12:
Log4j is a vulnerability in Java, which is not used by motionEye. motionEye uses Python for its backend and regular HTML/CSS/JavaScript for its frontend, so there is no chance it can be affected by Log4j 😉. I'll mark this as closed.

closed 2022-03-12 by MichaIng