motioneye-project/motioneye #2270: Any chance that the video playback will ever work on Android

feraldruid22, 2022-01-13:
The android app is very useful and the real time feed works fine.
However I have tried to play back the videos on multiple Androids running on phones and tablets and it's the same issue just a gray square the video won't play.

Can you please fix this in the new versions?

Thanks

starbasessd, 2022-01-13:
First, there isn't an Android app. MotionEye/motionEyeOS uses a WebGUI that is accessible in the web browser. Any specific 'android app' was written by another person, and you should ask them for support.
Second, you need to see what video codecs are supported by the android app / android phone, and use that when saving the videos. If your phone/tablet doesn't support H.264/OMX (.mp4) then try the other options available (there are 12)
I do not have issues with mine. 
Are you accessing via HomeAssistant or another similar system or direct at port 8765? If using HA or other, try accessing directly.

feraldruid22, 2022-01-14:
Well its really not hard to find it if you enter Motioneye to the search in the appstore thats pretty much the only app which comes up by FutureJJ.
As this is a small project you two should just cooperate to make the best out of it.
I bet my phone and tablet supports H264 since they are new devices running the latest Android.

starbasessd, 2022-01-14:
Try installing VLC video player to make sure the codec is available.

closed 2022-01-14 by starbasessd