motioneye-project/motioneye #2271: Preserve Movies Not working

BBGamingNL, 2022-01-13:
Have a raspberry pi 4 with installation on Docker. and tried the preserve movies setting. Tried a week and custom in 3 days. But both don't work

starbasessd, 2022-01-13:
What is your base OS?
Which Docker environment?
What is your docker run command?
What volumes are shared?

starbasessd, 2022-01-14:
Which docker install system? Docker.io or Docker-ce?
Try starting the container using the command in the wiki:
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
As to not 'sharing', you can redact, as long as you are consistent,
but, because how 'sensitive' volumes are in docker, you may be missing something (a needed volume), have the wrong -v path:path, or something else that I would need to see to help.
Your choice.

BBGamingNL, 2022-01-15:
> Which docker install system? Docker.io or Docker-ce? Try starting the container using the command in the wiki: https://github.com/ccrisan/motioneye/wiki/Install-In-Docker As to not 'sharing', you [...]

I'm using Docker ce. I used the Install in docker link and copied and pasted it. and when i put in command line docker volume ls i get local and then a code (8cf021e819ea4222b1eda0c0541d2519ffa46ff18c2f0f56982f5e98ef1c2985)

starbasessd, 2022-01-15:
Creating a Pi3B+, RaspberryPiOS (2021/05/07 - pre Bullseye and Legacy) Fully updated.
Will install docker-ce, and the docker image from WiKi with the command line
The path -v /var/lib/motioneye:/var/lib/motioneye  /var/lib/motioneye/CameraX should be where the movie files get saved. I will set my Preserve to 1 day and report back.

starbasessd, 2022-01-15:
Oh, one last question: Are you running motionEye docker container as pi or sudo?

BBGamingNL, 2022-01-15:
> Oh, one last question: Are you running motionEye docker container as pi or sudo? 

As sudo. En group docker, because then you don't have to put sudo in front of it

starbasessd, 2022-01-15:
I am confused by your answer. Are you running the docker run command from the Wiki as 
sudo docker run  (pi is not a member of the docker group)
or
docker run (pi is a member of docker group)
???

BBGamingNL, 2022-01-15:
docker run

starbasessd, 2022-01-15:
System set up. Running docker container as user pi.
Recording to local folder /var/lib/motioneye/Camera1 (as expected)
Will report tomorrow as to result.

BBGamingNL, 2022-01-18:
> System set up. Running docker container as user pi.
> Recording to local folder /var/lib/motioneye/Camera1 (as expected)
> Will report tomorrow as to result.  
> 

And what did it do by you?

starbasessd, 2022-01-18:
Will be checking this afternoon. Real life interfered with access to that docker image, unfortunately.

starbasessd, 2022-01-18:
As of right now, it is showing only yesterday and today in both the /var/lib/motioneye/Camera1 folder ( default paths ), and in the play movies icon in the Screen WebGUI after 49 hours uptime.
This was on a clean install of a VM in VirtualBox 6.1.30, Debian Bullseye (lite/Server/no Desktop) fully updated, with docker.io (version:  20.10.5+dfsg1, build 55c4c88) & docker-compose (unused), 4 Cores, 4GB RAM, 50GB HD assigned.
I ran the docker run command from the Wiki for motionEye [Docker Install](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker).
I added a single camera and set it to enable movies, Preserve 1 day. 
I set motion detection to .1% ( to make sure movies would be triggered ). 
Worked as expected.

starbasessd, 2022-01-18:
I will leave running another 24 hours to check again tomorrow, at about this time.

starbasessd, 2022-01-18:
Similar setup on Pi3B+, with RaspberryPiOS Buster fully updated, docker-ce (version 20.10.12, build e91ed57)
I ran the docker run command from the Wiki for motionEye [Docker Install](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker).
I added a single camera and set it to enable movies, Preserve 1 day.
I set motion detection to .1% ( to make sure movies would be triggered ).
Worked as expected. Only files from yesterday and today showing in WebGUI and /var/lib/motionEye/Camera1. Total uptime for Pi3B+ is 3 days, 3 hours, 26 minutes.
Will continue this machine for 24 hours more for test.

starbasessd, 2022-01-19:
Both my Pi3B+ (docker-ce) and Debian Bullseye (docker.io) VM have the same results today. They have /var/lib/motioneye/Camera1/2022-01-18 & /var/lib/motioneye/2022-01-19 folders and files for both. Both properly deleted the 2022-01-17 folders and files, as set by Preserve 1 day. 
I am unable to verify your issue in either setup, using the standard paths used in the [Docker Install](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker) suggested docker run command.

BBGamingNL, 2022-01-19:
Okay, will recreate the image. And will close this post

starbasessd, 2022-01-20:
For grins and giggles, I left both machines up one more day. Both are showing as expected, (as of 20 Jan) only 19 Jan and 20 Jan files.

closed 2022-01-20 by BBGamingNL