motioneye-project/motioneye #2272: Problem installing on Debian Bullseye

kennyheimbuch, 2022-01-15:
Getting some errors installing.  
Step #3 = dh-python: breaks python. Could be [this issue.](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=972726)

Step #4, tried both pip install motioneye and pip2 install motioneye, both lead to error:

`sudo pip2 install motioneye
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Collecting motioneye
  Using cached motioneye-0.42.1.tar.gz (758 kB)
Collecting tornado<6,>=3.1
  Using cached tornado-5.1.1.tar.gz (516 kB)
Collecting jinja2
  Using cached Jinja2-2.11.3-py2.py3-none-any.whl (125 kB)
Collecting pillow
  Using cached Pillow-6.2.2-cp27-cp27mu-manylinux1_x86_64.whl (2.1 MB)
Collecting pycurl
  Using cached pycurl-7.43.0.5.tar.gz (216 kB)
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-YSUAZS/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-YSUAZS/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-181ome
         cwd: /tmp/pip-install-YSUAZS/pycurl/
    Complete output (11 lines):
    Traceback (most recent call last): 
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-YSUAZS/pycurl/setup.py", line 961, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/tmp/pip-install-YSUAZS/pycurl/setup.py", line 623, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-install-YSUAZS/pycurl/setup.py", line 101, in __init__
        self.configure()
      File "/tmp/pip-install-YSUAZS/pycurl/setup.py", line 238, in configure_unix
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.`

and then step #5 leads to directory not found:

`sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory`

Wondering if this is down to the documented Debian issue linked above of if I did something wrong. (did an install on KDE Neon (so Ubuntu 20.04 last month, went flawlessly)

Running: Debian Bullseye (MX Linux 21), Python 2.7.18/Python 3.9.2.

starbasessd, 2022-01-15:
dh-python is not part of motionEye install for Debian Bullseye. I don't know anything about MX Linux so don't know if that is your root cause. The other errors are because motionEye and its dependencies did not fully install.
I verified motionEye install on Debian 11 Bullseye on 22 Dec 2021. I can do so again if you feel the need.

kennyheimbuch, 2022-01-15:
This is the part that's odd to me because MX basically is just Debian stable under the hood (Bullseye now), with some extra tools added, but uses 100% stable repos.

Is there a way to backtrace the cause of dependencies not installing?

starbasessd, 2022-01-15:
dh-python and python-is-python2 are fatally incompatible. 
Because python-is-python2 cannot install and its modules and dependencies. 
You might try purging dh-python, and starting the motionEye install over, but I don't know what in MX Linux depends on dh-python.

kennyheimbuch, 2022-01-15:
Bingo! apt-get purge dh-python and then going through it all again from the beginning worked!

Thanks for your help!

starbasessd, 2022-01-15:
Close if resolved.

closed 2022-01-15 by kennyheimbuch