motioneye-project/motioneye #2273: Login info

recfrf44, 2022-01-16:
how to make the code of the main page not visible when authorizing in the background?

starbasessd, 2022-01-16:
What are you referring to? Screen Shot maybe?

recfrf44, 2022-01-16:
and another html code, I want to see only the authorization window, without unnecessary information

[image]

starbasessd, 2022-01-16:
What is 'unnecessary info'? You could re-write the python login code, I suppose, if you don't want to see which motionEye you are logging into, You can remove the temperature wherever you added it, etc.

recfrf44, 2022-01-16:
any information other than the authorization window is insecure. Authorization must go first, and then the rest

starbasessd, 2022-01-16:
No different than any other website anywhere on the internet.

closed 2022-01-16 by starbasessd