motioneye-project/motioneye #2274: second motioneye server - connection to camera lost

caslor, 2022-01-16:
Hi to all friends, i have one issue and would like your advice
i have setup in a raspberry pi 4  4Giga  a docker a motioneye :
motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Debian 9.12

and as cameras i am using a  esp32cam modules  (4 of them)  
everything work fine....
i want to use this setup to detect motion and send me warnings and also i have make it to video stream the cams 

in a second installation in a orange pi zero  pc  i have install the same : 
motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Debian 9.12

i want to use the second setup just to record only continuously the cams video.

The problem is that when i am adding in the second server   the cams through 
Camera Type : Network camera 
URL : http://192.168.1.45:8081
etc..

i have the cameras added but no video...   everytime i am starting them i have for some seconds Green screen and after a Grey screen with message  : CONNECTION TO CAMERA LOST SINCE .........

When i am adding the cameras  as a : Simple MJPEG camera  then everything work fine , i am receiving the video but the problem is that Motioneye doesnt allow the video capture mode in the MJPEG type of cameras

Thanks in advance

starbasessd, 2022-01-16:
Try taking the video stream from the Pi, rather than the esp32cam directly. It has been noted on many sites that some esp32cams (esp32s in general) only support one connection at a time.
So, you have the first Pi (ip_address) looking at cam1
It is Camera1 on it
You would have the second pi look for the stream at http://ip_address:8081

starbasessd, 2022-01-16:
Can you pull the feed from the Camera directly? Most esp32cams only support a single feed, but a few do support multiple connections.

caslor, 2022-01-17:
Hi. i am not using the esp32cam to stream in both...only my main raspberryPi receive the video feed from the esp32

  my setup is like this

Esp32cam -- wifi--->  Raspberry Pi ----stream video---> Orange Pi zero

starbasessd, 2022-01-17:
Try:
&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; RPi
Cam\<
&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; Orange Pi
To see if your esp32cam can support more than 1 feed.
The other option is to put the Orange Pi before the RPi in the sequence, take the stream feed from the Orange Pi.

caslor, 2022-01-17:
i have tried to connect the same cam in both  raspbbery & orange pi but only one each time could connect.
i tried it again now the same...

unfortunately neither the

closed 2022-01-17 by caslor