motioneye-project/motioneye #2277: IP Camera viewer for the Raspberry Pi

rtek1000, 2022-01-17:
Hello,

Does MotionEyeOS have any function to view the camera of another MotionEyeOS?

I found this other project, but I haven't tested it yet:

> 
Use your Raspberry Pi as advanced IP camera viewer. The list of supported IP cameras is endless as Camplayer makes use of the RTSP and HTTP protocols for streaming. Especially the RTSP protocol is supported by nearly all big brands out there.

https://github.com/raspicamplayer/camplayer

starbasessd, 2022-01-17:
both motionEye and motionEyeOS can act as hubs for other motionEye/motionEyeOS machines. Either as add camera, network camera, or remote motioneye camera, or simple mjpeg camera. motionEyeOS and motionEye support RTSP/RTMP cameras, too. You can use the [iSpy Connect](https://www.ispyconnect.com/cameras) database to find URLs for most brands

JamBalaya56562, 2026-11-10:
Closing, as this was answered: motionEye and motionEyeOS can show the cameras of another one, added as a Remote motionEye Camera (with that server's Client Secret), a Network Camera or a Simple MJPEG Camera.

closed 2026-11-10 by JamBalaya56562