motioneye-project/motioneye #2278: Grey camera feeds, VNC can't display after enabling legacy cameras

jedtrently, 2022-01-17:
OS: Debian 11 Bullseye
ME version: latest

So....I installed ME, working fine except for completely block grey camera feeds. Did some reading and it turns out there's some issues with Bullseye and ME and the libraries blah blah blah

So I enable legacy camera support and now VNC doesn't work...

Any ideas if it's something to do with ME and bullseye compatbility?

starbasessd, 2022-01-17:
You don't state what kind of cameras. If you are using CSI (ribbon cable PiCams) you probably need to use Buster Legacy or the Buster image from 5/10/2021. If you are using network cameras it should work fine. The issue is noted as the 3rd bullet point in the Bullseye instructions.

jedtrently, 2022-01-17:
> You don't state what kind of cameras. If you are using CSI (ribbon cable PiCams) you probably need to use Buster Legacy or the Buster image from 5/10/2021. If you are using network cameras it should work fine. The issue is noted as the 3rd bullet point in the Bullseye instructions.

Thanks buddy. Yeah, it's CSI ribbon cams. Any idea if this will ever be fixed for Bullseye? I guess i'll revert back to Buster for time being. Cheers for the help :)

starbasessd, 2022-01-17:
No they (RPiFoundation) have broken it intentionally and have no plans on putting it all back the way it was. There's a lot of upset for projects like this (besides python2 needs to be updated) and there's no support for the old libraries or apps. Motion-Project is having to do some stuff differently (the base under motionEye), and others.

jedtrently, 2022-01-18:
Issue caused by Bullseye incompatability with Motioneye. Reverted to Buster and issue fixed. Thank you

closed 2022-01-18 by jedtrently