motioneye-project/motioneye #2280: External PIR sensor trigger for recording

jedtrently, 2022-01-18:
I'd like to use a PIR sensor plugged into the GPIO to trigger video recording, instead of the software motion detection that is currently used. 

Does anyone know if this feature is available, or anyone know how to do it?

I've seen a few other request and threads talking about it but to my knowledge nobody has quite done it yet.

Thanks

starbasessd, 2022-01-18:
Can you? Yes. 
Has something been written 'officially'? No
There are MANY PIR modules out there, using GPIO SPI, UART, I2C, and other methods of communicating. If you are using motionEye on another OS, you can find some hints [here](https://maker.pro/raspberry-pi/tutorial/how-to-interface-a-pir-motion-sensor-with-raspberry-pi-gpio)
You could then have it trigger a motion event direct to motion:
> curl localhost:7999/1/action/eventstart

and when PIR no longer senses movement
> curl localhost:7999/1/action/eventend

jedtrently, 2022-01-18:
Rightyho. Thank you for that. So how exactly would I 'trigger a motion event direct to motion' ?

starbasessd, 2022-01-18:
"Rightyho. Thank you for that. So how exactly would I 'trigger a motion event direct to motion' ?"
In whatever program you use to monitor the PIR sensor, or script you write, when the program or script senses the movement, it would need to use the command I gave to start recording, and when it stops detecting, use the second command. 
That is beyond the support I can give as I do not have any PIR Sensors to test with, let alone without knowing what PIR sensor you are going to use. If you want to have it done for you, ask someone who knows Python, or NodeRed, or maybe check out something like HomeAssistant or similar programs.

closed 2022-01-18 by starbasessd