motioneye-project/motioneye #2282: High resolution images

ToViRu, 2022-01-20:
Hi, 

I'm using RpiZeroW2 and I'm not able to use the full resolution (3280x2464) of the Camera Board v2. I want to record still images at an interval of 1-2 hours, but no images are saved unless the resolution of the camera is decreased bellow1920. 

This problem is quite similar to the issues #1764. Was it resolved?

Viktor

starbasessd, 2022-01-20:
To answer the last question first, no, there wasn't any further discussion. There are limitations with motion/motionEye and the HQ/Hi Res cameras.
If you are not using any of the functionality of motionEye (motion detection, FNC, etc), it may be that the better solution might be to run RaspberryPiOS Bullseye, and a cron job at whatever interval you want and the new [libcamera](https://www.raspberrypi.com/news/bullseye-camera-system/) functions.

ToViRu, 2022-01-20:
Thanks a lot for the info and suggestions!

closed 2022-01-21 by ToViRu