motioneye-project/motioneye #2283: Question: Performance on PI02W

MajorMadeV1, 2022-01-21:
First of all thanks to jawsper and starbasessd for your contributions to the Raspberry Pi Zero 2 W.
I do own the old Pi Zero W and the Pi Zero 2 W running with MotionEyeOS.

My question is, have you guys experienced any performance boost with the new Pi Zero 2 W?
Personally, I could not boost my performance with the new Pi and I would like to hear from you if you noticed any changes?
I do not expect a leap jump in performance as the memory is basically the same, yet a bit faster but the CPU should make a difference, no?

starbasessd, 2022-01-21:
If it was for a single, network camera, or a PiCam (set to FNC) I would expect an improvement over the PiZeroW, but for multiple cameras, due to the memory limitation, I wouldn't expect much difference. The CPU is much better, but there isn't much improvement on GPU or the rest of the system. You could try upping the [gpu](https://www.raspberrypi.com/documentation/computers/config_txt.html#gpu_mem) memory a bit (default is 128) to 192 to see if that helps. The setting is in /boot/config.txt:
> gpu_mem=128
> gpu_mem_256=128
> gpu_mem_512=128

I do it before installing the SDCard in the PiZero...

MajorMadeV1, 2022-01-24:
> I do it before installing the SDCard in the PiZero...
Any specific reason for that?

In about 3 weeks I`ll try and test some different configs on a testbench to see what the real-world performance differences will be. I am not sure this has been done. Certainly camera modules have been compared before but concerning the two pi performances and different configs, I could not find anything.

Thank you for your Input @starbasessd

starbasessd, 2022-01-24:
I do it because each camera in a multi-camera environment requires a certain amount of GPU usage for encoding and decoding video. I have come across errors that indicate a lack of gpu memory, and the slight increase on a PiZero/Pi1 helps it. Here is one link:
https://www.raspberrypi.com/documentation/accessories/camera.html#troubleshooting

WarmChocolateCake, 2022-01-24:
I'm planning to move my (several year old) Pi Zero based camera to a Pi Zero 2W

The current setup has the CPU pegged to near enough 100% (I'm using the Raspberry pi camera with 128MB GPU RAM + 10% overclocking, based on [ArchLinux for Arm](https://archlinuxarm.org))

I presumed that the Zero 2W would perform better purely due to having more CPU cores - but can motion/motioneye utilise more cores?

I saw that there were some issues with the camera firmware late last year (see: [Pi Zero 2 W camera troubles](https://forums.raspberrypi.com/viewtopic.php?t=323308) ), so I have been holding off until the dust settles, but I'd be interested to see your results @MajorMadeV1

JamBalaya56562, 2026-11-10:
Closing, as this was answered: the Zero 2 W's faster CPU helps with one camera, while its 512 MB of memory, the same as the Zero W, limits what several cameras gain.

closed 2026-11-10 by JamBalaya56562