motioneye-project/motioneye #2284: Unable to install on RPi 3b+ linux 11 bullseye

mudassar791, 2022-01-21:
Pkg is not upgrded for updated version of rpi 3b+.

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

starbasessd, 2022-01-21:
What do you mean pkg is not upgrded for updated version of rpi 3b+?
I have a Pi3B+ with Bullseye and motionEye installed (using these [docs](https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye)) and running. You cannot use MMAL (CSI ribbon cameras due to RPiFoundation breaking the raspivid and raspipic support structure) but network and USB cameras still work.

mudassar791, 2022-01-23:
[image]

mudassar791, 2022-01-23:
Every thing got completed but facing error in CP, so service canot be iniitated and started.

JohnZable, 2022-01-23:
I had the same issue.  You need to make sure both pip2 is installed, as pip3 installs in a different place, and you use sudo for the commands so you have write permission to the locations.

starbasessd, 2022-01-23:
Looks like you may be installing as a 'normal' user. You need to install as sudo, THEN follow the instructions [here](https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User) to run as a normal user.

starbasessd, 2022-01-23:
I just completed a new clean install using the following (from the Wiki for RaspberryPiOS-Bullseye) and it works as 'normal'. I then ran the Wiki for [Run as a Normal User](https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User)  and it is working 'normally'
[Install for RPiOS Bullseye plus motionEye.txt](https://github.com/ccrisan/motioneye/files/7920429/Install.for.RPiOS.Bullseye.plus.motionEye.txt)
Instructions for Install RPiOS-Bullseye updated with Testing date.

mudassar791, 2022-01-23:
Thankyou so much, just started from writing pi os on memory card then followed you, working perfectly.

closed 2022-01-23 by mudassar791