motioneye-project/motioneye #2286: Packages missing on Raspbian Bullseye

Manuel5cc, 2022-01-24:
I started to install Motioneye on a Raspberry Pi zero 2 W with the last version of Bullseye. Trying to install the first set of packages throws an error:

`Get:1 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian bullseye/main armhf shared-mime-info armhf 2.0-1 [697 kB]
Err:2 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian bullseye/main armhf libsnappy1v5 armhf 1.1.8-1
  503  Backend fetch failed [IP: 150.214.5.134 80]
Get:3 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian bullseye/main armhf libxcb-shape0 armhf 1.14-3 [102 kB]
Get:4 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian bullseye/main armhf libmicrohttpd12 armhf 0.9.72-2 [98.0 kB]
Fetched 897 kB in 55s (16.4 kB/s)
Unable to correct missing packages.
E: Failed to fetch http://ftp.cica.es/mirrors/Linux/raspbian/raspbian/pool/main/s/snappy/libsnappy1v5_1.1.8-1_armhf.deb  503  Backend fetch failed [IP: 150.214.5.134 80]
E: Aborting install.
`

starbasessd, 2022-01-24:
What did you find the error to be?

Manuel5cc, 2022-01-24:
> What did you find the error to be?

I couldn't find the cause of the issue so I tried to install the packages manually with "apt-get install" and they did without problems.
After that I continued following the instructions and everything worked.