motioneye-project/motioneye #2287: Ubuntu 22.04 python-pil only python3

reidjr, 2022-01-28:
I can no longer install motioneye following the wiki, as python-pil seems to be deprecated on 22.04 testing. All other packages seem to be installing as expected.

starbasessd, 2022-01-28:
In what OS?

starbasessd, 2022-01-28:
Checking.

starbasessd, 2022-01-28:
Also, you do realize the Daily Releases are Beta, that the final Desktop (and Server) releases aren't going to be for 3 more months, right?

starbasessd, 2022-01-28:
[EOL](https://endoflife.software/operating-systems/linux/ubuntu) for 20.04 LTS isn't until 4/2025, and 22.04 RELEASE DATE isn't until 4/21/2022

reidjr, 2022-01-28:
Hi, thanks for the reply.
Yes I realise its bleeding edge :-) but It will be the next major upgrade in 3-4 months on Ubuntu. If People have a working 20.04 install and its pseudo dedicated then they are good till 2025 as you say. I was running through a migration to see which of my "must have" services will fail on upgrade. As I have been using motioneye since at least 16.04, be sad to lose it :-(

The wiki suggests its 20.04 or newer and checked a month ago. As the last python-pil package was in 20.04, am I missing something here ? How is this working on 20.10-21.10 ?

starbasessd, 2022-01-28:
It's going to be later today, as the download is going at less than 800k/sec on a 400Mbit/sec connection...
While waiting I will re-try 21.10, but 20.10, 21.04 are already EOL'd and 21.10 EOL's in July.

reidjr, 2022-01-28:
Thanks, Sorry if I wasn't clear. I didn't want to use 20.10-21.10, I was just wondering if people have it working on any of those releases, where are they getting the python-pil package from ?

starbasessd, 2022-01-28:
Among things I will check. I haven't tested in a while, and need to anyway, I try to update the instructions with the last tested dates (especially Debian and RPiOS) but check anytime people ask for it, as that is how we found the updated process for python2-pip, and for a while, they kept moving the install file. 
_My_ 'new' policy for the testing _I_ post will be for supported OSs and the Install Wikis will reflect that, hopefully later today, just to help prevent any confusion. Debian Bullseye still has python2-pil, (as of 15 Jan, 2022) so Ubuntu should too. Have you tried the Debian Bullseye instructions?

reidjr, 2022-01-28:
Ok, we have contact :-). 
1) apt-get install python-pil -y  doesn't exist in Ubuntu repos  20.04 +
2) However its not needed, ( I think ) as

"pip2 install motioneye"
Requirement already satisfied: pillow in /usr/local/lib/python2.7/dist-packages (from motioneye) (6.1.0) ?

the final issue for me was I had not (re) installed motioneye as root, so /usr/local/bin/meyectl was not in the right directory, but the user directory.

So if the instructions are followed, and the error ignored (1) then it seems to work for me, which is great news. Hopefully support then till 2027 :-)

starbasessd, 2022-01-28:
I am currently updating (apt update && apt upgrade -y) 21.10. After I confirm instructions (and change as needed) I will also confirm on 22.04 (finally downloaded). This does NOT mean 22.04 is fully supported, though, until NET 27 April, 2022, as it is not officially released for Prime Time until then.

starbasessd, 2022-01-28:
Instructions updated, with clarifications. Please look them over and make any comments needed. Thank you for pointing out some improvements and clarifications needed.

reidjr, 2022-01-28:
Thanks for this. I ran a full upgrade from an image of 20.04 with my working motioneye config. Looks like it all works from a straight dist-upgrade (d) as well. So panic over, it still works in 22.04 :-)

As to the Instructions, that's great, it was just that one missing package, and removing that line avoids others stumbling as I did.

Thanks again

starbasessd, 2022-01-29:
What's up? Issue?

reidjr, 2022-01-29:
Perhaps I was a little hasty. The stored images (SNAPS) and the webfront end work as they should, but the displayed "real time" viewer on the web front end is distorted. Damaged Jpegs reported by chrome I think. I was rolling back and forward again 20.04 to 22.04 before I reported back properly.

starbasessd, 2022-01-29:
You are dealing with a beta OS and expect it to be 'perfect'? Try doing a backup in motionEye, then install a clean 22.04, a clean motionEye, then restore, and see if it works appropriately. It does for me.

reidjr, 2022-01-29:
Thanks for your help, good to know it works for you in the 22.04 beta, so its obviously at my end. I will give the clean install a try. I really appreciate the software, and the work that goes into it.

starbasessd, 2022-01-29:
Interesting, as I don't have that issue, but the version of motion for Jammy (22.04) is
4.3.2 and installs libmicrohttp12 0.9.75-3. Why it installs a known good motion, but a 'broken' libmicrohttpd12 is beyond me, with no way to downgrade...
Will update install document to not use 22.04, and because motionEye Development is now in limbo (as of 28 Jan 2022), may not be resolved.

reidjr, 2022-01-30:
Its a hack, but 21.10 has a package https://packages.ubuntu.com/impish/libmicrohttpd12 which is at libmicrohttpd12 (0.9.72-2). I can manually install that and fix the blocking. That works as I don't use anything ( a the moment ) that this package is a dependency for outside motion. Not to be recommended, but a possible work around for anybody stupid enough to be playing with 22.04 ;-). 

So download the 21.10 package, dpkg -i, and then sudo apt-mark hold libmicrohttpd12 so it is not upgraded.

Sorry to hear about this project moving to limbo status, its been going for a long time.

thanks again.

starbasessd, 2022-02-03:
RaspiOS 64 officially released, instructions updated as it default installs Libmicrohttp12_0.9.72.

MichaIng, 2022-03-12:
This should be obsolete with the switch to Python 3 (`dev` branch). I'll mark this as closed. Feel free to reopen if required.

closed 2022-03-12 by MichaIng

trbolexis, 2023-02-25:
Here is a link to the download: http://ftp.de.debian.org/debian/pool/main/libm/libmicrohttpd/libmicrohttpd12_0.9.72-2_amd64.deb

All should be performed as root `sudo` or `su`
1. `mkdir /tmp`
2. `cd /tmp`
3. `wget http://ftp.de.debian.org/debian/pool/main/libm/libmicrohttpd/libmicrohttpd12_0.9.72-2_amd64.deb`
4. `dpkg -i libmicrohttpd12_0.9.72-2_amd64.deb`
5. `apt-mark hold libmicrohttpd12`

> Its a hack, but 21.10 has a package https://packages.ubuntu.com/impish/libmicrohttpd12 which is at libmicrohttpd12 (0.9.72-2). I can manually install that and fix the blocking. That works as I don't use anything ( a the moment ) that this package is a dependency for outside motion. Not to be recommended, but a possible work around for anybody stupid enough to be playing with 22.04 ;-).
> 
> So download the 21.10 package, dpkg -i libmicrohttpd12_0.9.72-2_amd64.deb, and then sudo apt-mark hold libmicrohttpd12 so it is not upgraded.
> 
> Sorry to hear about this project moving to limbo status, its been going for a long time.
> 
> Thanks again.