motioneye-project/motioneye #2288: Problem (re)installing

speeskeek, 2022-01-28:
hi all,
been running for >5 years Motioneye no. very happy.
I run it on my openmediavault odroid home nas. was working very good.
I did an openmediavault upgrade today, and motioneye is broken.
was trying to reinstall but no luck - pillow is giving me a hard time.

`pip2 install pillow
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Collecting pillow
  Using cached Pillow-6.2.2.tar.gz (37.8 MB)
Building wheels for collected packages: pillow
  Building wheel for pillow (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-pOm8Zb/pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-pOm8Zb/pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-Cjjc6o
       cwd: /tmp/pip-install-pOm8Zb/pillow/
`

` apt-get install python-pillow
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python-pillow
`

debian 11.2
linux 4.14.222-odroidxu4
python3 --version : Python 3.9.2
python2 --version : Python 2.7.18
pip2 --version : pip 20.3.4 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)

speeskeek, 2022-01-28:
That's just a regular debian install..
it worked on that before.

using the wiki over here : https://github.com/ccrisan/motioneye/wiki/Install-On-Debian

starbasessd, 2022-01-28:
But the version of Debian you used (Bullseye), the instructions changed:
https://github.com/ccrisan/motioneye/wiki/Install-on-Debian-11-(Bullseye)

speeskeek, 2022-01-28:
oh ... tying that now... didn't see that one in the links or cant clock to that via the wiki ?

speeskeek, 2022-01-28:
great that works... thanks
please make the link to the debian 11 install a bit clear in the wiki ?

closed 2022-01-28 by speeskeek

starbasessd, 2022-01-28:
Yes, it's in the Wiki for the Installs on the right side, 2nd one down.

speeskeek, 2022-01-28:
if I go:
wiki - home - installation instructions - there's a list (debian, ubuntu, fedora, archi, ...) but not a link for 11..

starbasessd, 2022-01-28:
The link is on the right column but I did update the Main Installation page, too.