motioneye-project/motioneye #2289: libswscale.so.4

etamin419, 2022-01-28:
Hello
I have installed motioneye on the Debian 11 BullsEye Raspberry Pi, set up everything according to the instructions - **Install on Raspbian Bullseye**
But when i run `motion -h` I get an error:

> motion: error while loading shared libraries: libswscale.so.4: cannot open shared object file: No such file or directory

I tried to slip a file from debian 10 Busted from /usr/lib/arm-linux-gnueabihf/libswscale.so.4.2.100 and create symlink as libswscale.so.4
and get an error:

> motion: error while loading shared libraries: libswscale.so.5: wrong ELF class: ELFCLASS64

how to fix the error?

starbasessd, 2022-01-28:
Which version of motion? 
motion --version

etamin419, 2022-01-28:
motion 4.3.2

starbasessd, 2022-01-28:
Checking...

starbasessd, 2022-01-28:
Install RPiOS Bullseye 2021-10-30 Lite.
Copy wpa_supplicant.conf & ssh.txt files to /boot
Safely eject and install in RPi3B+, power on & connect via ssh
Fully update OS
Run install instructions from [here](https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye).
After install, tested for WebGUI, working
- - - - -
root@raspberrypi:/home/pi# motion -h
motion Version 4.3.2, Copyright 2000-2019 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen/Motion-Project maintainers

Home page :      https://motion-project.github.io/

usage:  motion [options]

Possible options:

-b                      Run in background (daemon) mode.
-n                      Run in non-daemon mode.
-s                      Run in setup mode.
-c config               Full path and filename of config file.
-d level                Log level (1-9) (EMG, ALR, CRT, ERR, WRN, NTC, INF, DBG, ALL). default: 6 / NTC.
-k type                 Type of log (COR, STR, ENC, NET, DBL, EVT, TRK, VID, ALL). default: ALL.
-p process_id_file      Full path and filename of process id file (pid file).
-l log file             Full path and filename of log file.
-m                      Disable motion detection at startup.
-h                      Show this screen.

Motion is configured using a config file only. If none is supplied,
it will read motion.conf from current directory, ~/.motion or /etc/motion.
- - - - -
Same result as user Pi. Sounds like a corrupt install for you. Try again starting at 
Step 2.
> sudo dpkg --purge motion
> wget https://github.com/Motion-Project/motion/releases/download/release-4.3.2/pi_buster_motion_4.3.2-1_armhf.deb 
> dpkg -i pi_buster_motion_4.3.2-1_armhf.deb

etamin419, 2022-01-29:
Thanks for the prompt replies
Understood the problem. I did a system update from 10 to 11. and before that, motion was also installed, which worked. When starting motion, it read the old configuration and took the outdated file from /usr/local/bin/motion. After deleting it, motion read the /usr/bin/motion file and started working

closed 2022-01-29 by etamin419