motioneye-project/motioneye #2291: Can't make my Raspberry Pi 4 2GB booting!

BirdCamp, 2022-02-15:
[image]

Bootloader 2022/01/25
Dev20201026
Replace the fixup4.dat and start4.elf files

No success at all.

Anybody any clue?

BirdCamp, 2022-02-15:
Even then, no success.

starbasessd, 2022-02-15:
RPi4-4GB Test Machine
pi@raspi1:~ $ sudo rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Tue 25 Jan 2022 02:30:41 PM UTC (1643121041)
    LATEST: Tue 25 Jan 2022 02:30:41 PM UTC (1643121041)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
No boot. Interesting. Checking other eeprom versions.

starbasessd, 2022-02-15:
I am not sure what is going on. I have 2 Pi4s, a 4GB and an 8GB, that have worked previously. The 4GB is now unable to boot motionEyeOS dev20201026. This will take some testing.

Gonioul, 2022-02-17:
I use motioneyeos on zero 1/2, raspbian on pi4, so I don't have the problem, but I would guess the shiny new bootloader with network install is the culprit.

starbasessd, 2022-02-17:
I went all the way back to the 4/2021 bootloader, and none of them seem to work. I am not sure what broke, but bringing them up to the 'default' (which doesn't have the 'shiny new network installer'), doesn't want to reverse. Still checking.  I freed up my Pi4-8GB for a bit:
- - - - -
pi@raspi5:~ $ sudo rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Thu 29 Apr 2021 04:11:25 PM UTC (1619712685)
    LATEST: Thu 29 Apr 2021 04:11:25 PM UTC (1619712685)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
with 
pi@raspi5:~ $ sudo rpi-eeprom-config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf124
- - - - -
Booted motionEyeOS dev20201026, no issues. moved the same card over to the Pi4-4GB with the 
pi@raspi1:~ $ sudo rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Fri 04 Feb 2022 12:03:08 PM UTC (1643976188)
    LATEST: Tue 25 Jan 2022 02:30:41 PM UTC (1643121041)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
- - - - -
Fails with the same SDCard, Will attempt to force back the BootLoader...
- - - - -
pi@raspi1:~ $ sudo rpi-eeprom-update
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Thu 29 Apr 2021 04:11:25 PM UTC (1619712685)
    LATEST: Tue 25 Jan 2022 02:30:41 PM UTC (1643121041)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
- - - - -
Nope, can't get it to boot the same known good (actually put it back in the Pi4-8GB and normal booted) SDCard. I don't know what broke, and putting it back to a known good rpi-eeprom doesn't fix it, either...

Ryu945, 2023-07-14:
There are two potential issues.  Try 1) before 2)

1)  You are not running the correct boot loader for MotionEyeOS.   Use the 4-29-21 version.   There is a program you can flash to a microsd card to change the bootloader version.  This guide talks about how to make one: https://pimylifeup.com/raspberry-pi-bootloader/

using the raspberry pi imager:  https://www.raspberrypi.com/software/

2)  I read somewhere that the new version of the Raspberry 4B boards are not compatible with MotionEyeOS.   Install a modern Raspberry PI OS and use this guide to install MotionEye.   I still haven't got this working myself so I can't help you with all the details but I got to see the GUI interface for MotionEye come up once.   https://jmichault.github.io/motioneye.eo-dok/en/instalado_en_debian/

Erfan-Fazeli, 2025-03-26:
Were you able to fix the problem? I also have the same error today and I couldn't fix the problem.

JamBalaya56562, 2026-11-12:
This is about booting motionEyeOS, which is a separate project ([motioneye-project/motioneyeos](https://github.com/motioneye-project/motioneyeos)); its last release is from 2020, and the dev20201026 build is the newest. motionEye itself installs on a current Raspberry Pi OS with the [README steps](https://github.com/motioneye-project/motioneye#installation). Closing here.

closed 2026-11-12 by JamBalaya56562