motioneye-project/motioneye #2294: Noob to docker

westilover, 2022-02-19:
Hi guys,
I currently use motioneye on an older Raspi - works like a charme but unfortunately the Raspi is to slow which causes lags ...

I now installed motioneye in a docker setting - unfortunately I'm completely new to docker. My problem is that the cameras that I configured in motioneye disappear after a reboot of the host system. How can I make that entries permanent?

Thanks for your support
Sascha

starbasessd, 2022-02-19:
What are you using for your docker run command?

westilover, 2022-02-19:
I use docker desktop running with windows 11 as OS . I click on Images - select ccrisan/motioneye - click on run - and select 8100 as Local host

so I'm sure Im missing something ...

as I said - unfortunately i'm a noob - shame on me *laughing*

Sascha

starbasessd, 2022-02-19:
I haven't tested on Win11, or even Windows 10, it will take me a bit to set all that up. 
If there are instructions to convert for a Linux command line to Windows docker desktop, I'd try to do the same setup as the instructions [here](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker#image-from-docker-hub).
In the mean time, I'll build a Win11 VM and see how I can destroy it...

starbasessd, 2022-02-19:
Sorry. 1st attempt at installing WSL 2 caused the VM to corrupt. New VM, attempting again.

starbasessd, 2022-02-19:
Apparently I do not have a

starbasessd, 2022-02-19:
I did not intend to close this. It's been a long day.

starbasessd, 2022-02-19:
OK, here goes: 
Add a Volume: motioneye 
Open a command line
enter:
docker run -d -p 8765:8765 --hostname="motioneye" -v motioneye:/etc/motioneye -v motioneye:/var/lib/motioneye ccrisan/motioneye:master-amd64
If you want to access the streaming ports directly, add -p 8081:8081 for the first camera, and -p 8082:8082 for the next... after the -p 8765:8765
This will set your docker-desktop up correctly and save the settings to survive reboots.

westilover, 2022-02-21:
WOW - thank you for GREAT support - works like a charm - now I only have to find out how to run that automatically  - i.e. every time my Win 11 machine starts up - start motioneye with the parameters you supported :-) There must be a way since my teslamate docker installation does that :-)

Again - thank you soooo much - I tried to do that for hours - and nothing worked - because I used the wron approach in the beginning (trying to start the system using the "click approach" in the docker App ...

Sascha

starbasessd, 2022-02-21:
You should be able to start and stop it using the Click options, now that it is set up correctly, it is more of how to get the 'persistent' volumes set up and used.
Win 11 is a different breed of cat, as is docker-desktop. Might want to get on a docker-desktop user forum and ask around there...

westilover, 2022-02-21:
Will do that - thanks again for the support :-)

starbasessd, 2022-02-21:
You're welcome. If working, please close.

closed 2022-02-25 by starbasessd