motioneye-project/motioneye #2296: Request to set date and time

patonadig, 2022-02-21:
Hello, this is a feature request. I have a standalone camera with a RPI zero W in which the WIFI mode is set to AP. I configure and retrieve the videos via the android app. The camera has no internet access and so the date / time is 1969. In order to set the date and time correctly I can SSH into it and configure it (I don't know any better way to do this). My main issue is that I might need to do this each time the system runs out of battery so I was looking for a simple way to do it. I can also create a script. I was wondering if it'd be possible to add a button in the UI to retrieve the system date and time and push it to the camera. 
Thanks!

starbasessd, 2022-02-21:
If you have a RTC hat, or GPS Dongle, it's easy to set the time to that. Anything else is problematic, as there isn't (currently) a way to set it in the WebGUI (which has already been asked for). 
If you have a RTC hat, instructions are [here](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-a-rtc-module)
If you have a GPS dongle, ask and I'll post instructions. (can be handled with python2.7 built in)

JamBalaya56562, 2026-11-10:
Closing, as setting the system clock is the operating system's job, which motionEye does not do; on motionEyeOS in access point mode that belongs to motionEyeOS (motioneye-project/motioneyeos). An RTC module, as starbasessd linked, keeps the time across power loss.

closed 2026-11-10 by JamBalaya56562