motioneye-project/motioneye #2297: Update dockerfile (necessary add support pantilthat library)

Govraskirill, 2022-02-22:
Hello, dear friends. I have the next problem, namely: 
I wanna using action buttons for ptz pro 2 camera. I found information, that i should: 
1. Create file left_1 (for left button) and put next code:
#!/bin/bash
/usr/bin/python3 /etc/motioneye/left.py
2. Create file left.py and put code:
#!/usr/bin/python

import time
import pantilthat

currentPos = pantilthat.get_servo_one()
newPos = currentPos +20
if newPos >= 80: newPos = 80
pantilthat.servo_one(newPos)
time.sleep(1)

But when i try to use pan tilt control i received the next message:

 WARNING: left_1: command has finished with non-zero exit status: 1
 WARNING: left_1: Traceback (most recent call last):
 WARNING: left_1:   File "/etc/motioneye/left.py", line 4, in <module>
 WARNING: left_1:     import pantilthat
 WARNING: left_1: ImportError: No module named 'pantilthat'

I assume that problem can be fix if I rebuild motioneye server image and put in dockerfile information about support pantilthat python lybrary. 
Please, give me advice how can i rebuild image so that fix this problem or other tips how to fix this problem

Govraskirill, 2022-02-22:
I tried ti add next command in Dockerfile:
RUN python -m pip install pantilthat
But this step can't help me and when i tried rebuild my image i received mistake (i did it on webstorm)

starbasessd, 2022-02-22:
You would need to add the proper lines in extra/Dockerfile if you are building a custom motionEye docker container, to include other python libraries. I updated the instructions in [here](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker#build-instructions).
I don't know if that command set will work for your camera, they appear to be for the Pan Tilt Hat, not for an ONVIF or similar network camera, though.

starbasessd, 2022-02-22:
BTW, you would also need to be sure that both python3 is installed in your docker container as the first section of code looks for python3, and you need to figure out if the second section of code can use python2 or python3, and fix the "#!/usr/bin/python" section to point to the correct python.
If you were to use the updated extra/Dockerfile listed in the instructions [here](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker) (at the bottom) find the line:
pip install /tmp/motioneye && \ 
and insert just below that as the next line:
pip install pantilthat && \
then try to rebuild the container. I don't guarantee it to work, but I've added other python libraries there like that.

starbasessd, 2022-02-22:
BTW, python-pantilthat library is built in to the motionEyeOS builder.

JamBalaya56562, 2026-11-10:
Closing with the answer: action button scripts run inside the container, so `pantilthat` has to be installed in the image for its `python3`. A small image `FROM motioneyeproject/motioneye` that installs pip as `docker/Dockerfile` does (it removes pip at the end) and then the module, run with the I2C device passed in (`--device /dev/i2c-1`), does that.

closed 2026-11-10 by JamBalaya56562