motioneye-project/motioneye #2298: can write movies but no webpage display

rob-miller, 2022-02-24:
Hi,

I installed 2022-01-28-raspios-buster-armhf-lite.zip on a Raspberry Pi B+ with original pi camera, then followed instructions at https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian to install motioneye.  

The only minor issue I detected was needing to add bcm2835-v4l2 to /etc/modules as described in the troubleshooting page, and then had to fix camera-1.conf to use /dev/video0 because the UI picked something else when I tried (perhaps useful to add 'get raspistill working first' to that install page?).

Now the camera light is on and I can motion-detect and capture 60s videos through the UI, but I get only get the camera-circle-slash icon on the webpage.  The only issue I can see in the log files is 'ECONNREFUSED', but from looking at other issues I think this just causes a retry until it goes through?

I attach syslog and output from 'meyectl statserver -d' from start through logging in as admin. Would be very happy and thankful if someone can highlight what I'm doing wrong?

[meyectl.log](https://github.com/ccrisan/motioneye/files/8134580/meyectl.log)
[meye-syslog.log](https://github.com/ccrisan/motioneye/files/8134582/meye-syslog.log)

starbasessd, 2022-02-24:
You are using http://ip_address:8765 for the WebGUI?

rob-miller, 2022-02-24:
Yes, i can see and use the web gui, just no image.

starbasessd, 2022-02-24:
Screenshot?

starbasessd, 2022-02-24:
How did you add the camera? It should be an MMAL camera, not the V4L2 camera. 
What does 
vcgencmd get_camera
report?

starbasessd, 2022-02-24:
What does http://ip_address:8081 show in you r web browser?
What does 
netstat -tulpn | grep 8081
report?
Are you running anything like HomeAssistant?

rob-miller, 2022-02-24:
I did add as the V4L2 camera, can't find where I saw to do that but did seem like was on the wiki.  changed:

    #videodevice /dev/video0
    mmalcam_name vc.ril.camera

in Camera-1.conf and restarted, same status except WebGUI shows MMAL camera / vc.ril.camera.  If enable movies in WebGUI still get a .mp4 file generated.

    pi@campi:/etc/motioneye $ sudo netstat -tulpn | grep 8081
    tcp        0      0 127.0.0.1:8081          0.0.0.0:*               LISTEN      5260/motion         
    pi@campi:/etc/motioneye $ 

    pi@campi:/etc/motioneye $ vcgencmd get_camera
    supported=1 detected=1
    pi@campi:/etc/motioneye $ 

> Are you running anything like HomeAssistant?

I'm not running HomeAssistant or any similar iot management system, not sure what this question would cover though.  I'm on same wifi as raspberry pi, normally using firefox but also tried safari for the screenshot.

> What does http://ip_address:8081 show in your web browser?

cannot connect.  Should have mentioned that earlier, but seems like motion working because of the .mp4 capture?

[image]

starbasessd, 2022-02-24:
Delete the camera
WebGUI, Settings, Click on Trashcan icon
Add camera
In Camera Drop Down, Click Add Camera
Add Local Camera MMAL (Not V4L2)
Something may be broken, building a setup, it may be a bit for me to fully test...

rob-miller, 2022-02-24:
Thank you for investigating!
Deleted camera, added MMAL, same status.
I'm away for a week from tomorrow, will pickup then if not later today.

starbasessd, 2022-02-24:
Yep, something is broken in your install, your camera isn't being shared to anything but localhost.
Yours:
pi@campi:/etc/motioneye $ sudo netstat -tulpn | grep 8081
tcp        0      0 127.0.0.1:8081          0.0.0.0:*               LISTEN      5260/motion         
Mine:
root@rpi-2298:/home/pi# netstat -tulpn |grep 8081
tcp        0      0 0.0.0.0:8081            0.0.0.0:*               LISTEN      4025/motion
I am able to connect from a Win10 with Chrome/Edge, and Firefox.  I can connect from a Pi4-4GB, also, to both :8765 and :8081

rob-miller, 2022-02-24:
cool, thanks for taking the time to find that.  will follow-up when I find
the issue.

rob-miller, 2022-02-24:
ok, got it:

at some point an error came up that there was no motion.conf in /etc/motioneye, and perhaps some other error stopped it from working further.  So I made a link to /etc/motion/motion.conf (which was *not* in the instructions) to get past that.  On removing that link, and deleting/re-adding the camera as MMAL, it is working again.
Sorry for the trouble, and thank you again.

closed 2022-02-24 by rob-miller

starbasessd, 2022-02-24:
The file for /etc/motioneye/motion.conf is NOT a link to /etc/motion/motion.conf, it is a 'stripped down' version. I would consider re-installing as that is NOT normal.