motioneye-project/motioneye #2300: motioneye not applying v4l2 config settings

jbs1, 2022-02-26:
I was trying to adjust the exposure settings for the camera I am using.

when doing
```
v4l2-ctl -c exposure_auto=1 (for manual mode)
v4l2-ctl -c exposure_absolute=20
```

the picture looks perfectly fine. but when i try to set the same setting in the "Video Device" menu in motioneye is simply doesn't change anything.

After changing the setting in the webinterface `v4l2-ctl -L` does not reflect the changes which i assume it should if they were applied correctly, right?

The other settings also seem to be kinda buggy. sometimes they apply, sometimes they don't and I can't really see a pattern here.
Anyone one have any idea what is wrong here?

starbasessd, 2022-02-26:
Exactly how and where in the WebGUI are you making your changes?

jbs1, 2022-02-26:
I am changing the setting here:

[image]

starbasessd, 2022-02-26:
How did you enable those settings?

jbs1, 2022-02-26:
when I added the camera in motioneye these settings just appeared by default. I didn't do anything special.
I just clicked "add Camera..." and selected the correct "Local V4L2 Camera"

starbasessd, 2022-02-26:
I built a new clean fully updated image, 
RaspberryPiOS 10, motion 4.3.2, and motionEye 0.42.1, 
added a V4L2 USB camera, and all I get is:
[image]
I've seen that menu with Fast Network Camera on motionEyeOS but not in motionEye.
I would have added the settings in the Extra Motion Options box.

jbs1, 2022-02-26:
hhm, in what format would you put the settings in the extra options?

I tried `video_params ID10094849=1,ID10094850=1515` as per https://motion-project.github.io/motion_config.html#video_params but that just gave me a

```
motion[944]: [0:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "video_params"
```

jbs1, 2022-02-26:
i didn't change anything. It still looks like before. i always had the "Extra Motion Options" it was just cut of on the screenshot.
but idk how to formal the options
[image]

jbs1, 2022-03-07:
so, I had some time to play around with this again.
I tried using the "Extra Motion Options".
First I tried using this format: `vid_control_params exposure_auto=1,exposure_absolute=20`
That seems to not be working as id does not even generate a log entry about setting the values.

Next I tried referencing the settings by id with `vid_control_params ID10094849=1,ID10094850=20`
This seems to actually do something because it generated this log:
```
Mar 07 10:28:55 wolpi motion[717]: [1:ml1:Squirrel Ca] [INF] [VID] vid_parms_parse: Parsing controls: ID10094849=1,ID10094850=20
Mar 07 10:28:55 wolpi motion[717]: [1:ml1:Squirrel Ca] [INF] [VID] v4l2_ctrls_set: Set control "Exposure, Auto" to value 1
Mar 07 10:28:55 wolpi motion[717]: [1:ml1:Squirrel Ca] [INF] [VID] v4l2_ctrls_set: Set control "Exposure (Absolute)" to value 20
```

And when checking `v4l2-ctl -L -d /dev/video0` it actually changed the values.
However it led to a very overexposed image.

So what I am thinking. I guess it actually switched the camera to Manual exposure mode but then for the exposure value it took the default value of 166 which would certainly lead to overexposure.
(Default Value as defined by `v4l2-ctl -L -d /dev/video0`)
No Idea why this is happening.

As a side note I noticed another weird behaviour during testing. Every config value has a default value as shown in `v4l2-ctl -L -d /dev/video0`. For all values it seems when switching to a different value works fine and it will generate the respective logentries and it will show up in v4l2-ctl but then trying to switch back to the default value through the sliders or "Extra Optiones" it will just straight up ignore the setting and not even generate the log entry to set the value.

As for the Exposure issue, I will probably just write a startup script that will set the value on every reboot or something.