motioneye-project/motioneye #2301: problems with python

1uk, 2022-02-26:
Hello, just tried to install motioneye on ubuntu 18.04.6 lts.
Here is the error I got:
```
Feb 26 14:37:06 server meyectl[31241]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 298, in make_media_folders
Feb 26 14:37:06 server meyectl[31241]:     camera_ids = config.get_camera_ids()
Feb 26 14:37:06 server meyectl[31241]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 321, in get_camera_ids
Feb 26 14:37:06 server meyectl[31241]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 409, in get_camera
Feb 26 14:37:06 server meyectl[31241]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 1872, in _set_default_motion_camera
Feb 26 14:37:06 server meyectl[31241]:   File "/usr/lib/python2.7/posixpath.py", line 68, in join
Feb 26 14:37:06 server meyectl[31241]:     if b.startswith('/'):
Feb 26 14:37:06 server meyectl[31241]: AttributeError: 'int' object has no attribute 'startswith'
Feb 26 14:37:06 server systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Feb 26 14:37:06 server systemd[1]: motioneye.service: Failed with result 'exit-code'.

```

Could I even skip the installation process, and use the raspberry image for a tower pc?

1uk, 2022-02-26:
I followed the instructions from your link - I think.
I will try to use the docker image, don't know how that works with ports and stuff though.
I don't know about docker.

starbasessd, 2022-02-26:
The docker image depends on how many cameras you are planning on using. 
For each camera, you would add a line:
-p 8081:8081
and increment for the number of cameras. IOW, if you have 3 cameras, you would have:
-p 8081:8081
-p 8082:8082
-p 8083:8083
in your docker run command
I am about 1/2 way through the install in an Ubuntu 18.04 VM.

starbasessd, 2022-02-26:
I just confirmed the install instructions are good on Ubuntu 18.04 (Server) with no issues (updated).
Did you try to install/run as a different user? I've never come across that error before.
Did you try to move any folders or other items?

starbasessd, 2022-02-26:
I use Debian 11 Bullseye on a Lenovo M73 TFF PC as my 'production' motionEye Server, I have used Ubuntu 18.04, 19.04, 20.04, 21.04, and 22.04 on both similar hardware and VMs. I use RaspbianOS Buster on Pi3B, 3B+, 4B-4GB and 4B-8GB for testing (Issues with Bullseye & PiCams). I've used or am using motionEyeOS on everything in the Pi Lineup (Pi1, Pi2, PiZero, PiZeroW, Pi3B, Pi3B+, Pi4B-4GB, and Pi4B-8GB) mostly motionEyeOS 20200606 & dev20201026. I have not tested on any other SBCs.
I do recommend Debian 10 or 11, Ubuntu 20.04, 22.04, Docker on almost any OS, and motionEyeOS on Pi hardware.

starbasessd, 2022-03-01:
What's your status?

MichaIng, 2022-03-12:
This is when a camera name is purely numeric. There is a fix merged already, and a better fix suggested, see the two linked PRs.

MichaIng, 2022-03-13:
Solved in `dev`, marking as closed. If there is still a need for a further Python 2 release. Feel free to open a PR with the same change against `python2` branch. Otherwise I suggest to wait for the Python 3 release to get rid of the Python version which is EOL since more than two years.

closed 2022-03-13 by MichaIng