motioneye-project/motioneye #2302: ONVIF control

Govraskirill, 2022-03-02:
Good evening, dear friends. I have the next problem, namely: I tried to connect several IP-cameras to motioneye. I did it. Next step I want to control this devices (HD-IP Alert Tracking PTZ Camera).
I used next instruction: https://www.instructables.com/MOTIONEYE-NVR-PANTILT/
My code for Up button: 

!/usr/bin/env python3

#Pan Tilt application for ONVIF devices

from time import sleep
from onvif import ONVIFCamera

XMAX = 1
XMIN = -1
YMAX = 1
YMIN = -1

def perform_move(ptz, request, timeout):
    # Start continuous move
    ptz.ContinuousMove(request)
    # Wait a certain time
    sleep(timeout)
    # Stop continuous move
    ptz.Stop({'ProfileToken': request.ProfileToken})

def move_up(ptz, request, timeout=1):
    print('move up...')
    request.Velocity.PanTilt._x = 0
    request.Velocity.PanTilt._y = YMAX
    perform_move(ptz, request, timeout)

def move_down(ptz, request, timeout=1):
    print('move down...')
    request.Velocity.PanTilt._x = 0
    request.Velocity.PanTilt._y = YMIN
    perform_move(ptz, request, timeout)

def move_right(ptz, request, timeout=1):
    print('move right...')
    request.Velocity.PanTilt._x = XMAX
    request.Velocity.PanTilt._y = 0
    perform_move(ptz, request, timeout)

def move_left(ptz, request, timeout=1):
    print('move left...')
    request.Velocity.PanTilt._x = XMIN
    request.Velocity.PanTilt._y = 0
    perform_move(ptz, request, timeout)

def continuous_move():
    mycam = ONVIFCamera('192.168.1.201', '80', 'admin', '123456', '/etc/onvif/wsdl/')
    # Create media service object
    media = mycam.create_media_service()
    # Create ptz service object
    ptz = mycam.create_ptz_service()

# Get target profile
    media_profile = media.GetProfiles()[0]

    # Get PTZ configuration options for getting continuous move range
    request = ptz.create_type('GetConfigurationOptions')
    request.ConfigurationToken = media_profile.PTZConfiguration._token
    ptz_configuration_options = ptz.GetConfigurationOptions(request)

    request = ptz.create_type('ContinuousMove')
    request.ProfileToken = media_profile._token

    ptz.Stop({'ProfileToken': media_profile._token})

    # Get range of pan and tilt
    # NOTE: X and Y are velocity vector
    global XMAX, XMIN, YMAX, YMIN
    XMAX = ptz_configuration_options.Spaces.ContinuousPanTiltVelocitySpace[0].XRange.Max
    XMIN = ptz_configuration_options.Spaces.ContinuousPanTiltVelocitySpace[0].XRange.Min
    YMAX = ptz_configuration_options.Spaces.ContinuousPanTiltVelocitySpace[0].YRange.Max
    YMIN = ptz_configuration_options.Spaces.ContinuousPanTiltVelocitySpace[0].YRange.Min

#PT control

    move_up(ptz, request)

if __name__ == '__main__':
    continuous_move()

But when I try do something, nothing work. Question: maybe motioneye server doesn't support ONVIF-command and so I should use for this goal other NVR-servers like Shinobi and other?

starbasessd, 2022-03-02:
How to set up on screen buttons: [here](https://github.com/ccrisan/motioneye/wiki/Action-Buttons)
1st, the command script should work on its own, first. IOW, you should be able to run the script on its own and the camera would move. If you have camera 1, and the script set up for /etc/motioneye/up_1 you should be able to run the command:
> /etc/motioneye/up_1 

and camera 1 should execute the up movement.
Once that is working, then the button in the screen view should work.
Post your script for the camera and we might be able to see where you are having issues.

Govraskirill, 2022-03-02:
Of course, I try to describe. 

Firstly, I created file on the next directories: /etc/motioneye/up_3 (because it’s third ip camera with Id =3)

Secondly, I did:

sudo pip3 install suds-py3
sudo pip3 install git+https://github.com/miuhaki/suds-passworddigest-py3.git
sudo pip3 install onvif-py3

Thirdly, I put the next code on up_3:

#!/usr/bin/env python3

#Pan Tilt application for ONVIF devices

from time import sleep
from onvif import ONVIFCamera

XMAX = 1
XMIN = -1
YMAX = 1
YMIN = -1

def perform_move(ptz, request, timeout):
    # Start continuous move
    ptz.ContinuousMove(request)
    # Wait a certain time
    sleep(timeout)
    # Stop continuous move
    ptz.Stop({'ProfileToken': request.ProfileToken})

def move_up(ptz, request, timeout=1):
    print('move up...')
    request.Velocity.PanTilt._x = 0
    request.Velocity.PanTilt._y = YMAX
    perform_move(ptz, request, timeout)

def move_down(ptz, request, timeout=1):
    print('move down...')
    request.Velocity.PanTilt._x = 0
    request.Velocity.PanTilt._y = YMIN
    perform_move(ptz, request, timeout)

def move_right(ptz, request, timeout=1):
    print('move right...')
    request.Velocity.PanTilt._x = XMAX
    request.Velocity.PanTilt._y = 0
    perform_move(ptz, request, timeout)

def move_left(ptz, request, timeout=1):
    print('move left...')
    request.Velocity.PanTilt._x = XMIN
    request.Velocity.PanTilt._y = 0
    perform_move(ptz, request, timeout)

def continuous_move():
    mycam = ONVIFCamera('192.168.1.201', '80', 'admin', '123456', '/etc/onvif/wsdl/')
    # Create media service object
    media = mycam.create_media_service()
    # Create ptz service object
    ptz = mycam.create_ptz_service()

    # Get target profile
    media_profile = media.GetProfiles()[0]

    # Get PTZ configuration options for getting continuous move range
    request = ptz.create_type('GetConfigurationOptions')
    request.ConfigurationToken = media_profile.PTZConfiguration._token
    ptz_configuration_options = ptz.GetConfigurationOptions(request)

    request = ptz.create_type('ContinuousMove')
    request.ProfileToken = media_profile._token

    ptz.Stop({'ProfileToken': media_profile._token})

    # Get range of pan and tilt
    # NOTE: X and Y are velocity vector
    global XMAX, XMIN, YMAX, YMIN
    XMAX = ptz_configuration_options.Spaces.ContinuousPanTiltVelocitySpace[0].XRange.Max
    XMIN = ptz_configuration_options.Spaces.ContinuousPanTiltVelocitySpace[0].XRange.Min
    YMAX = ptz_configuration_options.Spaces.ContinuousPanTiltVelocitySpace[0].YRange.Max
    YMIN = ptz_configuration_options.Spaces.ContinuousPanTiltVelocitySpace[0].YRange.Min

#PT control

    move_up(ptz, request)

if __name__ == '__main__':
    continuous_move()

But when I pressed button up on the motioneye window nothing happened

starbasessd, 2022-03-02:
If you run the command in a terminal by itself:
> /etc/motioneye/up_3

what happens?

Govraskirill, 2022-03-03:
> If you run the command in a terminal by itself:
> 
> > /etc/motioneye/up_3
> 
> what happens?

When I ran command /etc/motioneye/up_3 i received next message in command line:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/suds/transport/http.py", line 78, in send
    fp = self.u2open(u2request)
  File "/usr/local/lib/python3.8/dist-packages/suds/transport/http.py", line 119, in u2open
    return url.open(u2request, timeout=tm)
  File "/usr/lib/python3.8/urllib/request.py", line 531, in open
    response = meth(req, response)
  File "/usr/lib/python3.8/urllib/request.py", line 640, in http_response
    response = self.parent.error(
  File "/usr/lib/python3.8/urllib/request.py", line 569, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.8/urllib/request.py", line 502, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.8/urllib/request.py", line 649, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 400: Bad Request

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/suds/client.py", line 652, in send
    reply = transport.send(request)
  File "/usr/local/lib/python3.8/dist-packages/suds/transport/http.py", line 178, in send
    return HttpTransport.send(self, request)
  File "/usr/local/lib/python3.8/dist-packages/suds/transport/http.py", line 86, in send
    raise TransportError(e.msg, e.code, e.fp)
suds.transport.TransportError: Bad Request

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 34, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 201, in wrapped
    return call(params, callback)
  File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 191, in call
    ret = func(**params)
  File "/usr/local/lib/python3.8/dist-packages/suds/client.py", line 559, in __call__
    return client.invoke(args, kwargs)
  File "/usr/local/lib/python3.8/dist-packages/suds/client.py", line 618, in invoke
    result = self.send(soapenv)
  File "/usr/local/lib/python3.8/dist-packages/suds/client.py", line 664, in send
    result = self.failed(binding, e)
  File "/usr/local/lib/python3.8/dist-packages/suds/client.py", line 726, in failed
    raise Exception((status, reason))
Exception: (400, 'Bad Request')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/etc/motioneye/up_3", line 78, in <module>
    continuous_move()
  File "/etc/motioneye/up_3", line 46, in continuous_move
    mycam = ONVIFCamera('192.168.1.201', '80', 'admin', '123456', '/etc/onvif/wsdl/')
  File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 267, in __init__
    self.update_xaddrs()
  File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 283, in update_xaddrs
    capabilities = self.devicemgmt.GetCapabilities({'Category': 'All'})
  File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 36, in wrapped
    raise ONVIFError(err)
onvif.exceptions.ONVIFError: Unknown error: (400, 'Bad Request')

Govraskirill, 2022-03-03:
> > If you run the command in a terminal by itself:
> > > /etc/motioneye/up_3
> > 
> > 
> > what happens?
> 
> When I ran command /etc/motioneye/up_3 i received next message in command line:
> 
> Traceback (most recent call last): File "/usr/local/lib/python3.8/dist-packages/suds/transport/http.py", line 78, in send fp = self.u2open(u2request) File "/usr/local/lib/python3.8/dist-packages/suds/transport/http.py", line 119, in u2open return url.open(u2request, timeout=tm) File "/usr/lib/python3.8/urllib/request.py", line 531, in open response = meth(req, response) File "/usr/lib/python3.8/urllib/request.py", line 640, in http_response response = self.parent.error( File "/usr/lib/python3.8/urllib/request.py", line 569, in error return self._call_chain(*args) File "/usr/lib/python3.8/urllib/request.py", line 502, in _call_chain result = func(*args) File "/usr/lib/python3.8/urllib/request.py", line 649, in http_error_default raise HTTPError(req.full_url, code, msg, hdrs, fp) urllib.error.HTTPError: HTTP Error 400: Bad Request
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last): File "/usr/local/lib/python3.8/dist-packages/suds/client.py", line 652, in send reply = transport.send(request) File "/usr/local/lib/python3.8/dist-packages/suds/transport/http.py", line 178, in send return HttpTransport.send(self, request) File "/usr/local/lib/python3.8/dist-packages/suds/transport/http.py", line 86, in send raise TransportError(e.msg, e.code, e.fp) suds.transport.TransportError: Bad Request
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last): File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 34, in wrapped return func(*args, **kwargs) File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 201, in wrapped return call(params, callback) File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 191, in call ret = func(**params) File "/usr/local/lib/python3.8/dist-packages/suds/client.py", line 559, in **call** return client.invoke(args, kwargs) File "/usr/local/lib/python3.8/dist-packages/suds/client.py", line 618, in invoke result = self.send(soapenv) File "/usr/local/lib/python3.8/dist-packages/suds/client.py", line 664, in send result = self.failed(binding, e) File "/usr/local/lib/python3.8/dist-packages/suds/client.py", line 726, in failed raise Exception((status, reason)) Exception: (400, 'Bad Request')
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last): File "/etc/motioneye/up_3", line 78, in continuous_move() File "/etc/motioneye/up_3", line 46, in continuous_move mycam = ONVIFCamera('192.168.1.201', '80', 'admin', '123456', '/etc/onvif/wsdl/') File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 267, in **init** self.update_xaddrs() File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 283, in update_xaddrs capabilities = self.devicemgmt.GetCapabilities({'Category': 'All'}) File "/usr/local/lib/python3.8/dist-packages/onvif/client.py", line 36, in wrapped raise ONVIFError(err) onvif.exceptions.ONVIFError: Unknown error: (400, 'Bad Request')

I assume maybe problem in login, password for onvif. I found information that login/password for onvif may be different from the user/password to access the web interface

starbasessd, 2022-03-03:
I do not know if that is the case, but until that command works on it's own, it won't work as a button, either. I would suggest contacting your camera manufacturer, or an ONVIF support forum for the script you used.

Govraskirill, 2022-03-03:
> I do not know if that is the case, but until that command works on it's own, it won't work as a button, either. I would suggest contacting your camera manufacturer, or an ONVIF support forum for the script you used.

Thanks so much for your helping. Have a nice day

closed 2022-03-03 by starbasessd