motioneye-project/motioneye #2303: error " docker: failed to register -sh: 15: layer" during installation of motioneye in docker on usb drive

wielensw, 2022-03-02:
below the command I gave and the result....

docker is on usb drive connected to raspberry 4b. Who can help?

-----------------------

$ sudo docker run --name="motioneye" -p 8765:8765 -p8081:8081 --hostname="motioneye" -v /et                                                                                                                                                  c/localtime:/etc/localtime:ro -v /etc/motioneye:/etc/motioneye -v /var/lib/motioneye:/var/l                                                                                                                                                  ib/motioneye --restart="always" --detach=true --device=/dev/video0 ccri$ san/motioneye:mast                                                                                                                                                  er-armhf
Unable to find image 'ccrisan/motioneye:master-armhf' locally
-sh: 13: $: not found
$ -sh: 14: Unable: not found
$ master-armhf: Pulling from ccrisan/motioneye
eaaa43e22ddc: Extracting  45.71MB/45.71MB
38b5f9df17bb: Download complete
5a36f2cd0f41: Download complete
957aa23b13ac: Download complete
1b23938bf6aa: Download complete
docker: failed to register -sh: 15: layer: symlink ../285ba1cb3339ec05fde94f4c05bf2701349em                                                                                                                                                  aster-armhf:: not found21bf6e905ae6568963d7a2ce919e/diff /srv/d
ev-disk-by-uuid-15DF-36CB/docker/overlay2/l/7U57GJT2U2YWP3U6RC4WF6D2D2: no such file or dir                                                                                                                                                  ectory.

wielensw, 2022-03-02:
Docker I installed via OMV-extra. Docker version 5:20.10.12~3-0~debian-buster

starbasessd, 2022-03-02:
You have a number of broken lines here. 
When you say "docker is on usb drive connected to raspberry 4b" what do you mean?
Which version of Proxmox and OMV? There are known issues with Proxmox. Ditto with OMV.

starbasessd, 2022-03-02:
My test:
Rpi4-4GB on 128GB USB Drive (boots from this drive)
RaspiOS Buster (all current updates)
Install OMV using these instructions:
> https://pimylifeup.com/raspberry-pi-openmediavault/

Installed Docker using these instructions:
> https://forum.openmediavault.org/index.php?thread/40489-how-to-prepare-omv-to-install-docker-applications/

Installed motionEye using these instructions:
> https://github.com/ccrisan/motioneye/wiki/Install-In-Docker

(sudo'd because I was in a hurry)
pi@raspberrypi:~ $ sudo docker run --name="motioneye"     -p 8765:8765     --hostname="motioneye"     -v /etc/localtime:/etc/localtime:ro     -v /etc/motioneye:/etc/motioneye     -v /var/lib/motioneye:/var/lib/motioneye     --restart="always"     --detach=true     ccrisan/motioneye:master-armhf
Unable to find image 'ccrisan/motioneye:master-armhf' locally
master-armhf: Pulling from ccrisan/motioneye
eaaa43e22ddc: Pull complete
38b5f9df17bb: Pull complete
5a36f2cd0f41: Pull complete
957aa23b13ac: Pull complete
1b23938bf6aa: Pull complete
Digest: sha256:1d7abea3e92e8cc3de1072a0c102a9d91ef2aea2a285d7863b1a5dadd61a81e1
Status: Downloaded newer image for ccrisan/motioneye:master-armhf
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
30725abe60934298a860461c0c6b3cbdbac60f722cb2972fab41daa5f58a5f67
pi@raspberrypi:~
The WARNING line is 'normal', noted on the [Install In Docker](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker) page, and can be ignored due to using docker-ce instead of docker.io installation.
Was able to log in and add a network camera no problem.

wielensw, 2022-03-04:
followed the instructions but one thing I do different:
-I install raspiOS legacy light (buster 2022-01-28)  on the micro sd, ran all updates and upgrades
-I install OMV 5 via putty
-I install docker via OMV-extra's with storage location: /srv/dev-disk-by-uuid-15DF-36CB/DATA/docker

but still I get the same error. Can you try they I did to see if it works?

wielensw, 2022-03-04:
Are you willing to have a look with me via screensharing via teams? would help a lot spend allready 2 days fixing this....:(

starbasessd, 2022-03-04:
Building another setup right now. As soon as I get it up, I will let you know.
I do not do direct access or screen sharing support, too much liability for my wallet.

starbasessd, 2022-03-04:
How do you have your USB drive formatted?

starbasessd, 2022-03-04:
NTFS, FAT32/EXFAT/, EXT2, EXT4 or something else?

starbasessd, 2022-03-04:
32 GB SD card imaged with Buster 2022-01-28 Lite (enabled SSH & Wifi for first boot)
installed, connected via PuTTY
ran:
> sudo apt update && sudo apt upgrade -y
> sudo raspi-config
> &nbsp; &nbsp; updated password and hostname
> &nbsp; &nbsp; updated locale and timezone

rebooted
connected via PuTTY
ran command to install OMV v5.6.26-1
> wget -O - https://github.com/OpenMediaVault-Plugin-Developers/installScript/raw/master/install | sudo bash

Opened Web page for OMV maintenance, 
clicked System, OMV-Extras, Docker, Docker, Install
docker installed
Without changing default path clicked save
(changing the docker storage DOES NOT change where motionEye stores video files or information, that is handled by the -v volume settings in the docker run command)
Connect via PuTTY
run command:
docker run --name="motioneye" \\
&nbsp;&nbsp;&nbsp;&nbsp;-p 8765:8765 \\
&nbsp;&nbsp;&nbsp;&nbsp;--hostname="motioneye" \\
&nbsp;&nbsp;&nbsp;&nbsp;-v /etc/localtime:/etc/localtime:ro \\
&nbsp;&nbsp;&nbsp;&nbsp;-v /etc/motioneye:/etc/motioneye \\
&nbsp;&nbsp;&nbsp;&nbsp;-v /var/lib/motioneye:/var/lib/motioneye \\
&nbsp;&nbsp;&nbsp;&nbsp;--restart="always" \\
&nbsp;&nbsp;&nbsp;&nbsp;--detach=true \\
&nbsp;&nbsp;&nbsp;&nbsp;ccrisan/motioneye:master-armhf
- - - - -
pi@raspi6:~ $ sudo docker run --name="motioneye" -p 8765:8765 --hostname="motioneye" -v /etc/localtime:/etc/localtime:ro -v /etc/motioneye:/etc/motioneye -v /var/lib/motioneye:/var/lib/motioneye --restart="always" --detach=true ccrisan/motioneye:master-armhf
Unable to find image 'ccrisan/motioneye:master-armhf' locally
master-armhf: Pulling from ccrisan/motioneye
eaaa43e22ddc: Pull complete
38b5f9df17bb: Pull complete
5a36f2cd0f41: Pull complete
957aa23b13ac: Pull complete
1b23938bf6aa: Pull complete
Digest: sha256:1d7abea3e92e8cc3de1072a0c102a9d91ef2aea2a285d7863b1a5dadd61a81e1
Status: Downloaded newer image for ccrisan/motioneye:master-armhf
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
29427a3c2eca78a13dbf4a9415f91a5f5de1d5800ebfd822a422d73e9cf6e324
pi@raspi6:~ $
- - - - -
I am able to connect to http://ip_address:8765 and add a camera
No issues. Something is broken on your setup. Re-image and start over. Once you complete these steps, we can discuss how to get the data moved into the OMV storage.

starbasessd, 2022-03-04:
I am waiting for a freshly partitioned EXT4 USB drive to finish formatting, I will then join it to OMV to get the next stage working.

starbasessd, 2022-03-04:
I'm ready. 
BTW, what if you install docker, put in your /srv/dev-disk-by-uuid-15DF-36CB/DATA/docker, save, then run 
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-armhf
??? I just did that (moved docker storage to the /dev/sda /srv/dev-desk... /docker folder saved and ran the docker run command) and it worked, too...

wielensw, 2022-03-04:
USB drive =VFAT

Using the default docker storrage path works:

result

sudo docker run --name="motioneye" -p 8765:8765 --hostname="m             otioneye" -v /etc/localtime:/etc/localtime:ro -v /etc/motioneye:/etc/motioneye -             v /var/lib/motioneye:/var/lib/motioneye --restart="always" --detach=true --devic             e=/dev/video0 ccrisan/motioneye:master-armhf
Unable to find image 'ccrisan/motioneye:master-armhf' locally
master-armhf: Pulling from ccrisan/motioneye
eaaa43e22ddc: Pull complete
38b5f9df17bb: Pull complete
5a36f2cd0f41: Pull complete
957aa23b13ac: Pull complete
1b23938bf6aa: Pull complete
Digest: sha256:1d7abea3e92e8cc3de1072a0c102a9d91ef2aea2a285d7863b1a5dadd61a81e1
Status: Downloaded newer image for ccrisan/motioneye:master-armhf
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested edc43fb438547159c06464bc9c1a6c941ff2ee91b37329ab8e309454b178c708 docker: Error response from daemon: error gathering device information while adding custom device "/dev/video0": no such file or directory.

next thing I wanted to open motioneye via chrome via ip:8765 but a get a white page that site can not be reached. only fillling ip will end up in OMV offcourse. grrr

wielensw, 2022-03-04:
btw: this was the text in the omv extra installation window during installation of docker in the default storage location: some error during installtion it says in a line below.........

Docker storage :: /var/lib/docker
Agent port:: 8000
Web port:: 9000
Yacht port:: 8001
arch   :: armhf
option :: docker
state  :: install
extras :: 5.6.6
DNS OK.
Create override file ...
Reload systemd daemon ...
Get:1 file:/var/cache/openmediavault/archives  InRelease
Ign:1 file:/var/cache/openmediavault/archives  InRelease
Get:2 file:/var/cache/openmediavault/archives  Release
Ign:2 file:/var/cache/openmediavault/archives  Release
Get:3 file:/var/cache/openmediavault/archives  Packages
Ign:3 file:/var/cache/openmediavault/archives  Packages
Get:4 file:/var/cache/openmediavault/archives  Translation-en
Ign:4 file:/var/cache/openmediavault/archives  Translation-en
Get:3 file:/var/cache/openmediavault/archives  Packages
Ign:3 file:/var/cache/openmediavault/archives  Packages
Get:4 file:/var/cache/openmediavault/archives  Translation-en
Ign:4 file:/var/cache/openmediavault/archives  Translation-en
Get:3 file:/var/cache/openmediavault/archives  Packages
Ign:3 file:/var/cache/openmediavault/archives  Packages
Get:4 file:/var/cache/openmediavault/archives  Translation-en
Ign:4 file:/var/cache/openmediavault/archives  Translation-en
Get:3 file:/var/cache/openmediavault/archives  Packages
Ign:3 file:/var/cache/openmediavault/archives  Packages
Get:4 file:/var/cache/openmediavault/archives  Translation-en
Ign:4 file:/var/cache/openmediavault/archives  Translation-en
Get:3 file:/var/cache/openmediavault/archives  Packages
Ign:3 file:/var/cache/openmediavault/archives  Packages
Get:4 file:/var/cache/openmediavault/archives  Translation-en
Ign:4 file:/var/cache/openmediavault/archives  Translation-en
Get:3 file:/var/cache/openmediavault/archives  Packages
Ign:3 file:/var/cache/openmediavault/archives  Packages
Get:4 file:/var/cache/openmediavault/archives  Translation-en
Ign:4 file:/var/cache/openmediavault/archives  Translation-en
Get:3 file:/var/cache/openmediavault/archives  Packages
Get:4 file:/var/cache/openmediavault/archives  Translation-en
Ign:4 file:/var/cache/openmediavault/archives  Translation-en
Hit:5 http://httpredir.debian.org/debian buster-backports InRelease
Hit:6 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:7 http://packages.openmediavault.org/public usul InRelease
Hit:8 http://archive.raspberrypi.org/debian buster InRelease
Hit:9 https://openmediavault-plugin-developers.github.io/packages/debian usul InRelease
Hit:10 https://openmediavault.github.io/packages usul InRelease
Hit:11 https://download.docker.com/linux/debian buster InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  containerd.io docker-ce-cli
Suggested packages:
  cgroupfs-mount | cgroup-lite
Recommended packages:
  apparmor docker-ce-rootless-extras git libltdl7
The following NEW packages will be installed:
  containerd.io docker-ce docker-ce-cli
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/64.3 MB of archives.
After this operation, 274 MB of additional disk space will be used.
Selecting previously unselected package containerd.io.
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
[... 35 lines cut ...]
invoke-rc.d: initscript docker, action "start" failed.
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2022-03-04 22:12:11 GMT; 22ms ago
     Docs: https://docs.docker.com
  Process: 7662 ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock (code=exited, status=1/FAILURE)
 Main PID: 7662 (code=exited, status=1/FAILURE)
dpkg: error processing package docker-ce (--configure):
 installed docker-ce package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for systemd (241-7~deb10u8+rpi1) ...
Errors were encountered while processing:
 docker-ce
Failed to install docker.
Try fixing install...
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up docker-ce (5:20.10.12~3-0~debian-buster) ...
Successfully fixed docker install.
Installing local docker-compose 1.28.4...
Ready.

starbasessd, 2022-03-04:
Is http://ip_address_OMV_Server:8765 working or not?

wielensw, 2022-03-04:
Installed Portainer, I can open portainer via chrome: ip:9000. there I see the motioneye container. It has the "created" status.
When pushing restart of the continer or start button, i get message: request failed with status code 500.

starbasessd, 2022-03-04:
Portainer and the docker container of motionEye do NOT work together.

wielensw, 2022-03-04:
Is http://ip_address_OMV_Server:8765 is not opening. White window page that is can not reach site

starbasessd, 2022-03-04:
That error could be because of Portainer, or if you turned on the firewall, the firewall could be blocking it.

wielensw, 2022-03-04:
created a Stack in portainer and it worked. I am in the motioneye webinterface via IP:8765. !!!!!

Code for the stack:

---
version: "2"
services:
  motioneye:
    image: ccrisan/motioneye:master-armhf
    container_name: motioneye
    volumes:
      - /etc/localtime:/etc/localtime:ro #Timezone Config / Do Not Change
      - /srv/dev-disk-by-uuid-15DF-36CB/config:/etc/motioneye #Config Storage
      - /srv/dev-disk-by-uuid-15DF-36CB/motioneyedata:/var/lib/motioneye  #File Storage
    ports:
      - 8765:8765
    restart: unless-stopped

JamBalaya56562, 2026-11-10:
Closing, as the pasted command was split by line breaks (`-sh: 13: $: not found`), so the shell ran pieces of it, and it pulled the old `ccrisan/motioneye` image. The current image is `motioneyeproject/motioneye`, with the full command in `docker/README.md`.

closed 2026-11-10 by JamBalaya56562