motioneye-project/motioneye #2304: Performance rtsp

Tormund1985, 2022-03-03:
I find myself using motioneye along with CSI cameras such as the Raspberry IMX477 HQ along with Raspberry PI Zero.

However when using other PTZ cameras like Dahua or Hykvision the performance via RTSP is very bad... no motion detection is possible, in addition the image is pixelated.

I have checked with other software that this does not happen (Ispy), working really well.

I don't know why, I would like not to have to migrate to another server because with motioneye I really have everything I need. 

Can you tell me why Ispy captures well the rtsp broadcasts unlike motioneye?

Thanks in advance and thank you very much for your work.

Translated with www.DeepL.com/Translator (free version)

starbasessd, 2022-03-03:
I use Foscam PTZ cameras with RTSP and have no issues. I even use PiZero PiCams with motionEyeOS in Fast Network Camera mode, outputting RTSP, and they work fine. There are quite a number of cameras, though, that do not follow the standard that motion/ffmpeg were written to, and do not work well, if at all. I do not knowwhy one application works and another doesn't, other than the above.

Tormund1985, 2022-03-04:
I am working with Pi Zero 2W and Dahua rtsp streaming but the image is really bad and always pixeled, imposible to get motion detection. Between 0.2 and 0.7 fps under LTE connection. While with CSI cam's I get around 9fps.

starbasessd, 2022-03-04:
LTE? As in cell phone data?

Tormund1985, 2022-03-04:
> LTE? As in cell phone data?

exactly, with HAT SIM7600 from waveshare. The problem is not with the network itself but with the data acquisition by motioneye. Through VLC the rtsp link works perfect, but once you include it in motioneye the fps drop to max 3-4 and what is worse, pixelation of the image resulting in impossible motion detection.

starbasessd, 2022-03-04:
Are you running VLC on that same PiZero2?

Tormund1985, 2022-03-04:
yes, on the same pi Zero with VLC the video its exactly to Dahua streaming.

starbasessd, 2022-03-04:
What resolution is the camera sending, and what resolution is motionEye set to? Can you lower motionEye to 1280x720 (what I run my RTSP cameras at)?

Tormund1985, 2022-03-04:
It is set to 720p with the h264H codec and only 5fps for main stream.

starbasessd, 2022-03-04:
I don't know, then. My Camera connection:
[image]
My Streaming (what you see on the WebGUI):
[image]
My Movies settings:
[image]
I never use any of the movie formats that use the OMX. 
I get between 10 & 15 FPS across each of my 6 production cameras on a Gigabit Ethernet and WiFi (11ac) and USB2 network. 2 cameras are WiFi's, 3 are Ethernet, and 1 is USB2 (currently, I add and remove cameras for testing regularly) My Production server is a Lenovo M73-TFF, but I run motionEye and motionEyeOS on Pi1, PiZeroW (no where near the frame rate) Pi2 (slower framerate) Pi3B, 3B+, 4B=4GB, and 4B-8GB with similar frame rates. I run VMs on VirtualBox with many different OSs, and get similar frame rates.

starbasessd, 2022-03-04:
Why do you add your camera through Extra Motion Settings? If the camera is added in Add Camera, using your URL: 
> rtsp://192.168.1.108:554/cam/realmonitor?channel=1&subtype=0

and the username and password, you shouldn't need that line in Extra Motion.
Also, try RTSP/UDP when adding the camera

Tormund1985, 2022-03-04:
Just one question, why do you stream through the 8085 in addition to the necessary 8081? I understand that you get less fps this way than if it were deactivated.

Many thanks for your support, I don't have any Hat to test but next week I will try with your settings.

starbasessd, 2022-03-04:
Those settings are actually for camera-5 (out of the 6). Each camera is on it's own port (Cam1-8081, Cam2-8082... etc) camera-1 is streaming on 8081.

Tormund1985, 2022-03-07:
Hi again,

It's the same, the video pixelates when detect motion... In Dahua http streaming and VLC it's perfect

Tormund1985, 2022-03-17:
Hello,

This week I worked on a change in my environment.

I installed on a Raspberry pi zero 2 only a raspbian lite version, with NGrok and v4l2rtspserver services from a CSI camera and another rtsp streaming from a Dahua camera connected through LAN.

On the other hand on a Ryzen 7 server I installed motioneye and added both rtsp streams. well, now I get 30fps on each of them, with a great quality. That if, the CPU consumption is getting higher as I add cameras to the server. In this regard, if I had to buy a new server dedicated to manage the platform, which CPU would be the most suitable? I understand that GPU would not be necessary, and I wonder if an APU would be more suitable.

My intention is to set up an FTP server on this machine and manage about 50 rtsp transmissions.

Thanks in advance!

JamBalaya56562, 2026-11-10:
Closing, as this was a Dahua stream over LTE into a Pi Zero 2 W, where 0.2-0.7 fps points to the link and to Motion decoding the stream on that small CPU. The camera's low-resolution substream is the usual way to get detection going there. Please open a new issue with `motion.log` if a local stream is still that slow with a current release.

closed 2026-11-10 by JamBalaya56562