motioneye-project/motioneye #2306: Google will turn off plain smtp login - possible impact on motioneye-generated emails?

Merdk, 2022-03-07:
This is not an issue yet, but it might become one at the end of the month. Will this affect those of us using Gmail to send/store motion alerts? I received a message from google that on May 30 "​​Google will no longer support the use of third-party apps or devices which ask you to sign in to your Google Account using only your username and password." See their page [https://support.google.com/accounts/answer/6010255?rfn=1646361360139&anexp=nret-fa&pli=1&authuser=3#zippy=%2Cupdate-your-app-or-operating-system](url). Apologies in advance if this is baseless concern.

starbasessd, 2022-03-07:
May 30, not March 30, 2022.
I am curious myself, as Google flags my Amazon, Amazon Alexa, Debian Chromium, and motionEye, will any of them still work after May 30 2022, as it is not very clear on their site.

Merdk, 2022-03-08:
We'll have to wait and see! I noticed that some people sync motion media to gdrive, which for me would be an alternative to email (just not as handy as it would not create a push notification). But I wonder if authentication for that will also be affected in the same way.

starbasessd, 2022-03-08:
According to that [site](https://support.google.com/accounts/answer/6010255?rfn=1646361360139&anexp=nret-fa&pli=1&authuser=3#zippy=%2Cupdate-your-app-or-operating-system%2Cif-less-secure-app-access-is-on-for-your-account%2Cuse-an-app-password), if you use an app password (the Authorization Key), it (GDrive) should still be fine.
"[Use an App Password]

An App password is a 16-digit passcode that gives a non-Google app or device permission to access your Google Account. Learn more about how to [sign in using App Passwords](https://support.google.com/accounts/answer/185833)."

MichaIng, 2022-03-13:
I recently migrated our phpBB, Wordpress and Matomo successfully via:
- Enable 2FA for my Google account
- Create and apply a dedicated app passwords for each of the three applications

That worked very well without changing anything else than the password in SMTP settings. I use `LOGIN` method with SSL on port 465. No "insecure authentication" apps listen anymore in Google settings and safe for May 30 🙂.

Zombie-aaargh, 2023-11-26:
Has there been an update on this now that Google has made the change?

MichaIng, 2023-12-21:
Does it not work when you just create and use a dedicated application password?

Zombie-aaargh, 2023-12-22:
> 

I've already got an application password for using SMTP commands with Home Assistant, so I know that it's all set up properly. Which is why it's strange that I can't get it to connect through motion eye using the same log in details

MichaIng, 2023-12-22:
The idea behind application passwords is that you create one per application. So if this application gets compromised, you can remove the one password from your Google account without the need to change passwords everywhere else. This, apart of the common ability to limit access to particular resources only.

I am not sure whether/how this is enforced, e.g. based on some identified the client sends, but try to create and use a dedicated application password for motionEye.

Zombie-aaargh, 2023-12-22:
> The idea behind application passwords is that you create one per application. So if this application gets compromised, you can remove the one password from your Google account without the need to [...]

I'm not actually using Gmail, my isp uses their back end system, we only get one application password per email address and I use it on multiple devices to access my email. It's not a problem anywhere except here.

MichaIng, 2023-12-22:
Which SMTP settings and port do you use? Assure that TLS in enabled, as I am pretty sure (an hope) that Gmail is not accepting plain data transfer nowadays. If you use TLS already, please post the logs:
```sh
journalctl -u motioneye
```

One thing I recognised is that motionEye always starts an unencrypted connection and then uses STARTTLS when the "TLS" setting is enabled. This is nonsense IMO, as STARTTLS itself is in 99.9% of all cases: Either the SMTP server supports TLS (all serious servers do), then use TLS right from the start, or it does not, then I would never use this SMTP server in the first place.

Related code block: https://github.com/motioneye-project/motioneye/blob/ae03a4c/motioneye/sendmail.py#L44C12-L46

Should be changed:
```diff
--  conn = smtplib.SMTP(server, port, timeout=settings.SMTP_TIMEOUT)
    if tls:
--      conn.starttls()
++      conn = smtplib.SMTP_SSL(server, port, timeout=settings.SMTP_TIMEOUT)
++  else:
++      conn = smtplib.SMTP(server, port, timeout=settings.SMTP_TIMEOUT)
```
Or we change the option to allow only TLS or STARTTLS as alternatives, to never allow sending unencrypted data at all.

From the apps I use with Gmail, those either use port 465 with (true) TLS, or 587 with STARTTLS. Not sure whether it works when exchanging port and protocol. Since one usually expects "TLS" to actually mean TLS right from the start, usually at port 465, the current behaviour of motionEye could indeed lead to false settings.

Zombie-aaargh, 2023-12-23:
> Which SMTP settings and port do you use? Assure that TLS in enabled, as I am pretty sure (an hope) that Gmail is not accepting plain data transfer nowadays. If you use TLS already, please post the logs:

These are the instructions for setting up SMTP access form the ISP [Link](https://community.virginmedia.com/t5/Help/Virgin-Media-Email-settings/td-p/3166396)

These are the settings that are being used to access SMPT in Home Assistant which are running perfectly right now.

> notify:
  - name: email_notification
    platform: smtp
    server: "smtp.virginmedia.com"
    port: "465"
    username: "emailaddress@virginmedia.com"
    password: "password"
    sender: "emailaddress@virginmedia.com"
    sender_name: "Home Assistant"
    recipient: "emailaddress@virginmedia.com"
    encryption: tls

This is my motioneye configuration

[image]

I'm not sure what logs you might want or how to retrieve them

MichaIng, 2023-12-23:
Probably they do not support plain connection with STARTTLS, or only on the other port. Can you try port 587? At least Gmail accepts this for plain/STARTTLS.

MichaIng, 2023-12-23:
Can you try this:
```sh
apt install patch # assuming some DEB-based distro
cd /usr/local/lib/python3.*/dist-packages # assuming motionEye installed globally via pip
curl -sSf 'https://patch-diff.githubusercontent.com/raw/motioneye-project/motioneye/pull/2896.patch' | patch -p1
systemctl restart motioneye
```

Zombie-aaargh, 2023-12-23:
> Can you try this:

I'm using MotionEye through the Home Assistant app store.

Can I still use the above commands?

I have almost zero knowledge of understanding of anything beyond copy\pasting things into the command line.

MichaIng, 2023-12-25:
Oh, no the HA app is the old motionEye 0.42.1 and I am not sure how the files are distributed.

Probably run a `find . -name sendmail.py` from within the HA install/data directory and apply the patch from it's parent-parent directory (instead of of the `/usr/local/...` from my commands above) if it is called `motioneye`. Of not, you could still edit the file manually: https://github.com/motioneye-project/motioneye/pull/2896/files

closed 2026-11-08 by MichaIng