motioneye-project/motioneye #2323: build: pyproject.toml vs requirements.txt

cclauss, 2022-03-10:
@jmichault What is your recommendation for building this project?
* [ ] https://python-poetry.org
* [ ] https://flit.readthedocs.io
* [ ] Other?

The current GitHub Action is doing `pip install -r requirements.txt` but that does not take advantage of `pyproject.toml` and `setup.cfg`. How would you suggest that our GitHub Actions build the project to leverage these files?

jmichault, 2022-03-10:
No preference. I have made pyproject.toml and setup.cfg because it's recommended, but i have only used `python3 -m build` and `twine  upload`. Never used more advanced tools, you are more experienced than me for this part, so choose the most appropriate.

cclauss, 2022-03-10:
I have never used https://pypi.org/project/build so thank you for teaching me about it.

closed 2022-03-10 by cclauss