motioneye-project/motioneye #2327: ME does not start

fxtv1060, 2022-03-10:
I am running Armbian 21.08.8  with Linux 5.10.60-imx6 on a Cubox i. 

This morning I installed ME from the dev (?) repository. Everything went fine but I was unable to start ME: motioneye_init: command not found. Then I switched to the [Debian BE version](https://github.com/motioneye-project/motioneye/wiki/Installation). Everything went fine as well but I a still unable to access ME through my browser (http://XX.XX.XX.XX:8765) although Apache is running. 

I plan to connect two Eufy cameras and stored the recorded movies.

I quite sure I did something wrong...

François-Xavier

starbasessd, 2022-03-10:
Which set of install instructions did you follow from the [Installation](https://github.com/motioneye-project/motioneye/wiki/Installation) instructions here in the Wiki?

fxtv1060, 2022-03-10:
yep I followed "Install On Debian 11 Bullseye or Later](https://github.com/motioneye-project/motioneye/wiki/Install-on-Debian-11-%28Bullseye%29)". No error during the installation.

fxtv1060, 2022-03-10:
The issue is that I cannot use ME from a browser... I don't know how to investigate.

fx

starbasessd, 2022-03-10:
If you have a monitor and keyboard attached to the Cubox, run the command:
>sudo systemctl status motioneye

and post the results.
Also run the command:
>netstat -tulpn |grep 8765

and post the results.
Both should work in ARMBIAN or Debian.

fxtv1060, 2022-03-10:
Here are the results.

`root@cuplex:/var/log/motion# sudo systemctl status motioneye
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2022-03-10 17:55:20 CET; 33min ago
    Process: 9898 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
   Main PID: 9898 (code=exited, status=1/FAILURE)
        CPU: 287ms

Mar 10 17:55:19 cuplex systemd[1]: Started motionEye Server.
Mar 10 17:55:20 cuplex meyectl[9898]: Traceback (most recent call last):
Mar 10 17:55:20 cuplex meyectl[9898]:   File "/usr/local/bin/meyectl", line 5, in <module>
Mar 10 17:55:20 cuplex meyectl[9898]:     from motioneye.meyectl import main
Mar 10 17:55:20 cuplex meyectl[9898]:   File "/usr/local/lib/python3.9/dist-packages/motioneye/meyectl.py", line 28, in <module>
Mar 10 17:55:20 cuplex meyectl[9898]:     import settings
Mar 10 17:55:20 cuplex meyectl[9898]: ModuleNotFoundError: No module named 'settings'
Mar 10 17:55:20 cuplex systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Mar 10 17:55:20 cuplex systemd[1]: motioneye.service: Failed with result 'exit-code'.
`

`root@cuplex:/var/log/motion# netstat -tulpn |grep 8765
root@cuplex:/var/log/motion#`

starbasessd, 2022-03-10:
You're using Python3, and it's not supported yet. The installation instructions are for python2.7.
You cannot simply switch using the current format. 
If you cannot install python2.7 then it's not going to work for you.

starbasessd, 2022-03-10:
No need to. Walking through the instructions for Debian 11 on another machine right now, will confirm it all works and let you know.

starbasessd, 2022-03-10:
Install instructions correct, I just installed into Debian 11 Bullseye with no issues, and was able to connect to http://ip_address:8765 and added a camera. Instructions are correct. You must follow them step by step.

fxtv1060, 2022-03-11:
I am sorry but I didn't get what "Walking through the instructions for Debian 11 on another machine right now, will confirm it all works and let you know" means. Could you explcit a little bit.

fx

starbasessd, 2022-03-11:
I don't use my primary machine for testing installs.
I went step by step with the instructions from here:
https://github.com/motioneye-project/motioneye/wiki/Install-on-Debian-11-(Bullseye)
and confirmed each step worked (completed successfully)
I was able to confirm that doing so created a machine with a working motioneye that
responded to a web browser by going to http://the_ip_of_the_test_machine:8765
When I ran the tests I gave you:
>sudo systemctl status motioneye

I got a report with a line stating:
     Active: active (running) since Sat 2022-02-19 06:33:25 EST; 2 weeks 6 days
which says the motioneye service is working
>netstat -tulpn |grep 8765

I got a line stating:
tcp        0      0 0.0.0.0:8765            0.0.0.0:*               LISTEN
which shows it is correctly listening for connections
This confirms the instructions work. when you tried the command sudo systemctl status motioneye
it indicated it was trying to use Python3 instead or the required Python2, which I told you needed
to be corrected. Did you walk through the instructions again, and install step by step?

fxtv1060, 2022-03-11:
Many thanks. I used a new SD card and installed a fresh Armbian. The I followed the procedure step by step. 

Now it is working fine! 

Many thanks for you help and patience.

fx

closed 2022-03-11 by fxtv1060